bambulab/BambuStudio #2778: Modify Bambu Studio for easier manual filament color change

geeffland, 2023-10-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

geeffland, 2023-10-22:
Currently in Bambu Studio, if you have an AMS you can do a filament change which will allow you to color the layers on screen and easily verify that filament changes are marked for the correct layers.  If no AMS then filament change does nothing and you have to use "Pause".  This works except you then are not allowed to color the layers in Bambu Studio, making it very easy to pause in the wrong layer.  If there was a way to add a pause and also change colors in the layer on screen (like filament change does) then it would be easier to change filaments at specific layers and visually verify that the layers are correct.

Possible changes:
1) Bambu Studio allow use of Filament Change without an AMS and it automatically add a "pause" for non-AMS printers.  This way the colors could change on screen, printer file automatically change to "pause" (This might be the most intuitive option)
2) Allow color changes on layers @ pause layers, i.e. allow use of Pause as is, but add functionality to color layers different after the pause, similar to how Change Filament works.
3) Add option to "Filament Change" to allow for adding a pause also, or not
4) any other option that would allow visual checking of the location of a pause (other than just moving the preview slider up/down)

jollino, 2023-10-23:
Agreed, and the printer should do an unload/load cycle as a single group of action, instead of having to do it manually (like M600 on Prusa printers).

Another improvement, regardless of any of this, would be just sending a notification Bambu Handy when the printer reaches a pause, so that we know when it's time to go and do it. I think the P1 series doesn't come with a speaker (no idea about the X1), so right now we have to manually keep track of when the pause will place and make sure we're near the printer to do it.

bitflung, 2023-10-23:
this should also swap the actual filament profile - e.g. switching switching the temp, flow rate, etc to those of the next filament. i believe this already happens when you 'change filament' but want to be sure it isn't lost.

geeffland, 2023-11-14:
I agree with allowing changing the filament profile at the same time.

Not sure on Load/Unload... I could see the unload but after unload I remove the spool and swap spool with the new filament, so the Auto load would only work if waiting on a prompt to load

btittelbach, 2024-12-06:
fuck ack .. I fully expected this to work just like Prusa Printers .. and destroyed an 5 hour print as a result.

If I select  "color change" in the slicer and the printer does not see an AMS connected .. it should to a manual filament change.