bambulab/BambuStudio #2781: Can't print with Bambu PLA Matte profille

tjsnow, 2023-10-24:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

It works fine if I select other Matte PLA profiles, but specifically it seems to not work for this one. When I select the Bambu Lab Matte PLA profile in the device AMS settings, it looks like it sets it to ABS temporarily (for a few seconds) then updates, not sure if that is just load time or an actual problem.

### Printer

Bambu Lab P1S

### How to reproduce

1. load a non-bambu Matte filament into AMS
2. select the Bambu PLA Matte profile
3. Go back to model and resync filament
4. Set model to use that matte filament
5. Slice plate
6. Try to print

### Actual results

actual: system wont let me select that filament (even though it is what I sliced with) with a message stating "Note: Only the AMS slots loaded with the same material type can be selected

### Expected results

Expected: system will print using the profile

### Project file & Debug log uploads

[6-Shooter Rubber Band Gun.zip](https://github.com/bambulab/BambuStudio/files/13099465/6-Shooter.Rubber.Band.Gun.zip)
[image]
[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

tjsnow, 2023-10-27:
Additional Info:
I started rolling back version by version till it worked. It looks like this works in version 1.76.92, but is not correct in all version after that. I have also confirmed that even when I put a bambu lab brand matte filament in the AMS (using these later version), it too is seen as ABS, but works fine in 1.76.92.

walterwongbbl, 2023-10-31:
Hi. @tjsnow 
Have you used custom material presets or modified the filament type to ABS?

walterwongbbl, 2023-10-31:
@tjsnow 
We have also identified this issue, which appears to be related to using custom presets. We have made the necessary fixes and will be incorporating them in the next release.

tjsnow, 2023-10-31:
That was quick. No I do not use any custom presets. Glad you found it.
Looking forward to upgrading to the latest again.

TJ Snow
801-637-1890
***@***.***

walterwongbbl, 2023-11-06:
Hi. @tjsnow 
You mentioned you haven't used custom presets, could you provide your Studio Log file for us to confirm if it's the same issue causing this?

tjsnow, 2023-11-06:
I would be happy or upgrade and send that to you in a couple days. I am in
the middle of a large print order for someone and I need that profile to be
working, so I am using a downgraded version at the moment. I will try in
the next couple days to find time.
TJ Snow
801-637-1890
***@***.***

tjsnow, 2023-11-06:
I just upgraded between prints and the problem is still happening. I select
Bambu Matte Pla profile and it changes to the ABS profile, same when the
AMS automatically reads an matte pla filament. here are the logs.

walterwongbbl, 2023-11-07:
Hi.@tjsnow 
Please wait for the next version of Studio. 1.8.x.x

walterwongbbl, 2023-11-14:
@tjsnow 
We already have a new version, please use it.

tjsnow, 2023-11-20:
I keep checking, but no new version is available yet.

walterwongbbl, 2023-11-22:
@tjsnow 
https://model-file.bambulab.cn/private/CNc9045e9e208dc2/4179959/3mf/Metadata/plate_1.png?x-oss-traffic-limit=25165824&OSSAccessKeyId=LTAI5tFANCnfNNPNSSLjG97f&Expires=1700724163&Signature=oSAMEIj05sHL0AIGnyaqG0VrPPQ%3D

walterwongbbl, 2023-11-22:
@tjsnow 
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

tjsnow, 2023-11-25:
Thank you, I didnt realize it was a beta that i had to grab manually. i was looking for updates in the studio.

closed 2023-12-07 by walterwongbbl