bambulab/BambuStudio #2782: Part imports and slices with correct dimensions in Bambu studio but prints base layer with the wrong width

shuttle1987, 2023-10-24:
### Bambu Studio Version

v01.07.06.92-20230923002726

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 20.04.6 LTS

### Additional system information

Intel® Core™ i7-8750H CPU @ 2.20GHz × 12
GeForce GTX 1050 Ti Mobile

### Printer

Bambu Lab X1C with AMS

### How to reproduce

1. Create a new project
2. Import the 3mf file
3. Slice the plate
4. Print the plate
5. Printed part has incorrect dimensions on the base plate

### Actual results

Bottom plate is incorrectly printed with 40mm by 40mm dimensions while rest of the part dimensions are printed with correct dimensions.

The size of the edges in the square plate in the CAD drawings are specified as 45mm.

### Expected results

Bottom plate should be printed with 45mm by 45mm dimensions.

### Project file & Debug log uploads

[printing scale bug.zip](https://github.com/bambulab/BambuStudio/files/13110518/printing.scale.bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-10-25:
is the size in the studio is right?

shuttle1987, 2023-10-25:
The size of the base reports as 45mm by 45mm when viewed in the preview and as 45mm by 45mm after slicing the plate.

sodachen22, 2023-10-26:
@shuttle1987 I did not found problem when slicing and printing VaccuumAttachment-Body.3mf. It looks the issue may be related to the printer. Is there something wrong with your printer's accuracy if you only print a cube? If there is, please create a ticket and upload the log of the printer. https://wiki.bambulab.com/en/x1/troubleshooting/upload-log