bambulab/BambuStudio #2783: Replace painted color

Morialkar, 2023-10-24:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to replace a painted color. I tend to prepare a set painting of an object but like to put different variations. Sometimes, those objects have such particular variations in the model that the paint bucket does a poor job of even repainting an already painted area.

**Describe the solution you'd like**
I'd like to have, in the painting tool, a button to replace a painted color, this would give me a choice of the color I want to replace (say filament 1) and the target color (say filament 3) and would replace all spots painted with filament 1 with filament 3 painting automatically, allowing the creation of multiple variations of the same design with the same colors. Extra helpful with large multi-AMS paintings

**Describe alternatives you've considered**
I think a per object filament mapping setting could also solve this issue, but would be much more complex to handle for users I feel

**Additional context**
Add any other context or screenshots about the feature request here.

MadMax389, 2023-10-24:
I heartily second this request. I've had to completely repaint things when switching printers or loading up older profiles.

Shez-Make, 2024-11-28:
Put me down as a supporter of this feature. I would love to save time repainting a model I've done in the past. Unfortunately, saving the project doesn't save the colors and neither does saving as 3mf file.