bambulab/BambuStudio #2784: Problem printing PETG with PLA as interface layer with Grid infill

jorgeruimachado, 2023-10-24:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel(R) Core(TM) i9-9880H CPU @ 2.30GHz   2.30 GHz
RAM: 32Gb of RAM
Graphics: AMD Radeon Pro 5500

### Printer

Bambu Lab X1X

### How to reproduce

This is a sample file.
[File](https://drive.google.com/file/d/1zGJbpDOYHVP3W8LrOHHycNZFse0uElxO/view?usp=sharing
)
Changes to the 0,2 default configuration, on the file:
Material 1: Main Part material: Bambu lab normal PETG 
Material 2: Interface Layer: Bambu Lab Matte PLA (normal PLA still cause the issue)

Changes:
Support/raft interface: PLA ( Material 2)
Z-Top Distance: 0mm
Top interface spacing: 0mm

Door open

### Actual results

First and second layer of PETG will be ok
First and second interface layer, will be ok

After that, total caos, and difficult to describe. 

### Expected results

Looks like some serious under-extrusion.

### Project file & Debug log uploads

[Base_support.zip](https://github.com/bambulab/BambuStudio/files/13112844/Base_support.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-10-25:
if you don't mind, could you provide a photo of the print result? is there a ring stick on the surface?

jorgeruimachado, 2023-10-25:
This is a sample of the problem Best regards,Jorge Rui Silva

QingZhangBambu, 2023-10-30:
#2015 
there may be the same type of issue, I've talked to the concerned staff, and they are investigating it.

jorgeruimachado, 2023-10-30:
Just a heads up, I’ve manage to print a series’s of parts in PETG using PLA supports, but using Gyroid infill and purge values of plus 450.
So the caos seems to be related with the speed or geometry of the default infill type. 
hopes this helps pin point the problem