bambulab/BambuStudio #2785: This really SHOULD be packed in the AppImage! libwebkit2gtk-4.0.so.37

mprowe, 2023-10-24:
### Bambu Studio Version

v01.07.07.89

### Where is the application from?

Bambu Lab github releases

### OS version

KUbuntu

### Printer

Not relevent

### How to reproduce

1. Download the AppImage
2. Set the execute flag
3. ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage

### Actual results

```
mprowe@KUbuntu23-03:/AppImages$ ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
/tmp/.mount_Bambu_We144L/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
mprowe@KUbuntu23-03:/AppImages$ 

```

### Expected results

Program should run

### Project file & Debug log uploads

Not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jmarmstrong1207, 2024-04-13:
This needs to be fixed. AppImage was explicitly made so that dependency issues exactly like this WOULDN'T happen in any distribution.

skorokithakis, 2024-04-24:
Bambu Studio currently doesn't work for me at all, and I don't know how to work around this.

mansson, 2024-05-15:
I agree. This is urgent. Bambu Studio **Does Not Work** on Ubuntu 24.04

lanewei120, 2024-05-15:
please try the workaround method mentioned in https://github.com/bambulab/BambuStudio/issues/3973
currently

shebang79, 2024-05-16:
Disappointing a stable release was issued without this extremely basic and easily fixed critical issue being fixed. It'll be awhile before they can try again.

yunginnanet, 2024-07-07:
agree

lanewei120, 2024-07-17:
we have added a ubuntu image for 24.04 https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage

please use this one which is also a formal image

jmarmstrong1207, 2024-07-17:
> we have added a ubuntu image for 24.04 https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage
> 
> please use this one which is also a formal image
> 
> if there is no issue with this image, we will close this issue later

That AppImage still doesn't work on other distributions. I tested on Fedora. Error is the same:
`/tmp/.mount_Bambu_cnTjB4/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory`

To fix this, libwebkit2gtk-4.1 has to be packaged with the AppImage in order for it to work with every distribution and to avoid any further Linux incompatibilities overall

lanewei120, 2024-07-18:
> That AppImage still doesn't work on other distributions. I tested on Fedora. Error is the same: `/tmp/.mount_Bambu_cnTjB4/bin/bambu-studio: error while loading shared libraries: [...]

we tried to package those libraried into appimage before, but there is some strange issues on other platforms

on fedora, how about use fedora appimage on your side? https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_fedora-v01.09.03.50.AppImage
what is your fedora version?

MackBambu, 2024-07-18:
> That AppImage still doesn't work on other distributions. I tested on Fedora. Error is the same: `/tmp/.mount_Bambu_cnTjB4/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory`

I tested fedora 40 and it works fine, other versions are still recommended to use Fedora appimage.
> 
> To fix this, libwebkit2gtk-4.1 has to be packaged with the AppImage in order for it to work with every distribution and to avoid any further Linux incompatibilities overall

webkit2gtk has other dependencies, and packaging this library will cause the appimage to increase in size by 50M.

jmarmstrong1207, 2024-07-18:
> we tried to package those libraried into appimage before, but there is some strange issues on other platforms
>
> on fedora, how about use fedora appimage on your side? [...]

I am using Fedora 40 without libwebkitgtk installed, so it's predictable behavior.

> webkit2gtk has other dependencies, and packaging this library will cause the appimage to increase in size by 50M.

Unless the strange issues can be fixed, a larger size **is** expected for distro-agnostic formats such as AppImage, Flatpak, and Snap. Having multiple AppImages depending on the distribution is not the expected intent. For the implementation though, I recommend looking at the Flatpak version of Bambu Studio as that works perfectly without any additional packages being installed onto the machine. I believe Snap or Flatpak has a way to share libraries to reduce size though, and is more ideal and convenient than AppImage in my opinion, as they are both package managers

jmarmstrong1207, 2024-07-18:
> on fedora, how about use fedora appimage on your side? https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_fedora-v01.09.03.50.AppImage what is your fedora version?

Using Fedora 40, libsoup is missing now:
`/tmp/.mount_Bambu_VGhBLQ/bin/bambu-studio: error while loading shared libraries: libsoup-2.4.so.1: cannot open shared object file: No such file or directory`

jmarmstrong1207, 2024-09-18:
Is there any update with this? packaging the dependencies will be negligible in impact and will vastly solve installation issues. Even with the extra 50MB, it'd still be smaller than the Windows version

mprowe, 2024-10-24:
Any progress on these AppImages.....?
We (I) am getting farther and father behind! Not been able to move past 1.9.3.50.
I also note that there is an X1C Firmware update. Do we need to keep Studio and Hardware in-step?

Regards (but getting frustrated with your AppImages packaging), Martin

D0m-Sch, 2024-10-25:
I ran into the same issue with OrcaSlicer and QidiSlicer.
I found a "workaround" on the Playdate developer Forum they run into the same issue with their software.
The following commands worked for me with Orca- and QidiSlicer:

sudo add-apt-repository -y "deb http://gb.archive.ubuntu.com/ubuntu jammy main"
sudo apt update 
sudo apt install libwebkit2gtk-4.0-dev

Just replace libwebkit2gtk-4.0-dev with whatever 4.x version you need.

Here the link to the original Forum post:

https://devforum.play.date/t/issue-while-launching-the-playdate-simulator-on-ubuntu-24-04/17435

mprowe, 2024-10-25:
Thank you very much for the advice. However, as a mere "user" I have very little programming skills and what you suggest is not within my capabilities. Even if it was, are we missing the point! 

The Bambu Labs Studio is being shipped only "half-full" with required code missing. I think I should continue to request that Bambu provide a working program. And not expect users to try and add all the bits that Bambu have forgotten to package in the first instance. One possible way forwards maybe to slow down on the releases and ensure that more time is spent on testing that the next release actually WORKS?

Regards, M.

D0m-Sch, 2024-10-25:
I haven't used Linux in any form until quite recently and by copying those commands into a terminal I was able to circumvent the issue. 
However I absolutely get your point.
Either way the packages need to include the required dependencies, or Ubuntu needs to leave them available even after switching to 24.04.
In the OrcaSlicer thread someone suggested the same "fix" I did but I couldn't follow the instructions. That's why I provided the actual commands for the terminal.
Once again a circumvention, not an actual fix.

jlw-spqr, 2024-10-28:
Confirmed this still remains in  Bambu_Studio_ubuntu-24.04_PR-5090.AppImage when used against a Ubuntu 24.04 system.  Note:  I'm not sure it's an option but to avoid having users add and remove assorted package sources (out of date no less) you might consider statically linking the offending library into the appimage.  

Steps to reproduce:
1)  install OS, fully patch (sudo apt update && sudo apt dist-upgrade -y) 
2) download and run AppImage 

Details:
OS:  Ubuntu 24.04 fully updated as of 10/28
ERROR:  /tmp/.mount_Bambu_4Ln1oI/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory

probonopd, 2024-11-26:
I have made an experimental PrusaSlicer AppImage that bundles __all__ dependencies, including libwebkit2gtk:

https://github.com/probonopd/PrusaSlicer/releases/tag/mk1

For all users encountering issues related to libwebkit2gtk or other dependencies, maybe you can give that AppImage a try and share your experience (be sure to mention your Linux distribution and version). If beneficial, maybe an approach similar to what I have done there would be an option for BambuStudio as well.