bambulab/BambuStudio #2788: Camera window not working in device tab

tpchristian, 2023-10-25:
### Bambu Studio Version

1..7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Latest Patch Level

### Printer

X1 Carbon

### How to reproduce

This happens consistently for me.  The camera window is blank.  Pressing play button does nothing.  Pressing any of the small buttons above does nothing.  Except that one time pressing the live button crashed the application.

### Actual results

Nothing

### Expected results

Something, anything...  :)

### Project file & Debug log uploads

Happens with any/all projects.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-10-25:
They are not enabled before the printer is connected. So is the printer connected?

tpchristian, 2023-10-25:
Yes, I believe so.  It was originally not connected because I wasn't logged in.  I am able to send prints to the printer just fine.

bambu123, 2023-11-06:
@tpchristian Please refer to the following wiki for Studio logs to send to us. If there is a crash, please be sure to include the DMP files.

https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

tpchristian, 2023-11-06:
I have one DMP file so I am including that here.  Not sure which log to include, so just included a random one from the folder.  My camera window never displays the feed from the camera so maybe any of them will help.  Oh, I just now see that it won't let me upload the DMP file because it says it's too big.  Adding GD Link.

https://drive.google.com/file/d/1miNko3n67jH0E_SONw_7ljaxjwwXyLGY/view?usp=sharing

[debug_network_Fri_Oct_27_12_02_02.log](https://github.com/bambulab/BambuStudio/files/13267314/debug_network_Fri_Oct_27_12_02_02.log)

bambu123, 2023-12-22:
Sorry it's late, is your question still there?
Please refer to the wiki below to configure graphic settings.

https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view
Problem 5 - The program hung when loading the video stream! (Windows Only)

tpchristian, 2023-12-22:
Yes, I consistently have this problem.  I looked through the wiki link you pointed me to.  I looked through all of the items.  However specific to the one you pointed out, problem 5...  I don't have any discrete graphics.  I only have the Intel integrated graphics.  

I see that there is a new version.  I will give that a try and see if solves this.

bambu123, 2023-12-25:
Does it crash every time liveview is played?

tpchristian, 2023-12-25:
Yes, it does.

bambu123, 2023-12-26:
> Except that one time pressing the live button crashed the application.

Are you using an extended screen?

To analyze DMP files, we need you to provide the following DLLs on your system:

C:\Windows\System32\evr.dll
C:\Windows\System32\wmp.dll
C:\Windows\System32\msvcp140.dll
C:\Windows\System32\evr.dll
C:\Windows\System32\d3d9.dll

tpchristian, 2023-12-26:
I am using a single monitor.  I assume that is your question.  It is a widescreen 49".  I don't understand what you mean by "provide".  You want me to upload those here?

bambu123, 2023-12-26:
> I am using a single monitor. I assume that is your question. It is a widescreen 49". I don't understand what you mean by "provide". You want me to upload those here?

Yes, please zip them and upload here.

tpchristian, 2023-12-26:
Attached.  I assume you know that some of those files in your list were duplicates.
[bambu.zip](https://github.com/bambulab/BambuStudio/files/13768633/bambu.zip)