bambulab/BambuStudio #2789: Bug in filament selection in AMS

Rafimus, 2023-10-25:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab X1C

### How to reproduce

1. go to the printer tab
2. select filament in AMS
3. Set to GENERIC PLA
4. click to confirm
5. the filament box indicates ASA instead of PLA

### Actual results

Impossible to set Generic PLA in the menu. When selected, it changes to ASA automatically.

### Expected results

Generic PLA to be selected and possibility to set up the correct factors that go with it.

### Project file & Debug log uploads

[Emplacement_CVF-S_112mm_x3.zip](https://github.com/bambulab/BambuStudio/files/13162849/Emplacement_CVF-S_112mm_x3.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Rafimus, 2023-10-25:
Hello
Since the laste update of the software (for the AMS I assume, but maybe it's Bambu lab), when I'm selecting Generic PLA filament in the AMS, the software is turning it into ASA automatically and I can't change it. This is quite annoying as it won't use the filament when it's mixed with my Bambu brand filaments for multiple . I have to select Bambu PLA basic to be able to use it, but then the parameters and K factor are incorrect.
Has anybody ever heard of this problem before?
Thank you,
Best regards,
Raphaël Thirion

QingZhangBambu, 2023-10-30:
@Rafimus we've received severe issues with filament set on the device tab, the concerned staff are investigating it.

bam-winterfell, 2023-10-30:
Hi@Rafimus 
The user preset  uses the basic information of PLA material, and the "type" in "basic information" is modified to ASA. This resulted in AMS being unable to distinguish between ASA and PLA materials. You can use ASA related materials to create user presets instead of using PLA to create ASA user presets. The temporary solution to this problem is to delete the user presets that previously used PLA to create ASA.
We have noticed such issues and will fix them in the next version. Please update promptly. I apologize for the inconvenience caused to you.
Wishing you a pleasant life.

Hickinsauce, 2023-11-27:
Hello,
I was experiencing the exact same issue after using the Generic PLA base profile to create a PLA+ material profile. After this, all AMS slots set to Generic PLA would switch to PLA+ and thus would not allow me to print PLA since it recognized it as a different material. 

I can confirm that deleting the custom PLA+ profile I created has resolved the issue and everything is back to normal.

bam-winterfell, 2023-11-28:
@Hickinsauce The new version has fixed this type of issue, please update it.

Hickinsauce, 2023-11-28:
@bam-winterfell I am using the latest version available (1.7.7.89) and have confirmed the issue is still very much present in this version. It's fine as I know how to avoid it but it has not been "fixed".

bam-winterfell, 2023-11-29:
hi@Hickinsauce 
I'm sorry for not expressing myself clearly. These versions (01.08.56, 01.08.57, 01.08.62) have fixed this issue.The versions can be obtained from the "release page" of GitHub.And you can also wait for the official version to be pushed
[image]

bam-winterfell, 2024-02-04:
@Hickinsauce 
If you still have such issues, please report a new issue. Thank you

closed 2024-02-04 by bam-winterfell