bambulab/BambuStudio #2792: Unnatural power draw on macOS

GuzziRaz, 2023-10-26:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.1 (23B74)

### Additional system information

8-core intel, 32 GB RAM
intel UHD graphics
Radeon Pro Vega 20, 4 GB

### Printer

X1C

### How to reproduce

1. Start Bambu Studio. **Do nothing, just minimize the app**.
2. After a while, laptop fans runs hard (my laptop hardly ever needs the fan due to aluminum chassis).

### Actual results

3. Click battery/charge icon. It says "Using Significant Energy: BambuStudio.app".
4. Run system's Activity Monitor. Click "Energy" tab. Bambu Studio show a massive Energy Impact for no apparent reason. Click "CPU" tab. Bambu Studio show a ridiculous CPU load for not doing anything special and being minimized.
5. Switch to BambuStudio and select "Device" tab. Printer shows up.
6. System's Activity Monitor now show _even worse_ CPU usage, hogging all of a CPU core without video even being active mor any project loaded.

### Expected results

Bambu Studio should draw next to no resources when it's not performing any actual work.

Note: Problem is somewhat intermittent. It's very common to get stuck in this high power state after actually having a project loaded and device interaction, then stopping video and minimizing the app. But even in the best case, the app is **always** listed as "Using Significant Energy" when you click the battery/charge icon.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-12-25:
We are optimizing the performance of the application...

GuzziRaz, 2024-08-08:
This issue remains and the power draw is **ridiculous**. I don't know ANY other app that makes my macbook hot as h3ll even with the app not doing anything at all (no video active, for example). If running on battery, it's drained in no time.

spad, 2024-08-27:
Same here.... on MacBook Pro M3, SOnoma 14.6.1
In the preview panel, the GPU is allways to 100% even when minimized

fahaaaxiki, 2024-08-28:
@GuzziRaz Hi，we  are unable to reproduce this issue，could you provide a video showing how to reproduce it？

Haidiye00, 2025-07-22:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If there are still questions, we will analyze them again.

closed 2025-07-22 by Haidiye00