bambulab/BambuStudio #2793: Predictive nozzle temperature adjustment for sensitive materials

tsmith35, 2023-10-26:
After working with my X1C and P1P to print PETG, I have determined that the reason so many Bambu Lab users struggle to print with PETG is due to two things: a narrow usable temperature band for PETG and the high speeds of the Bambu Lab printers.

PETG goes from too cold (high viscosity) to too hot (low viscosity) very quickly. This becomes an issue when used with a fast printer; as the nozzle decelerates for corners, overhangs and other features, the nozzle heater does not respond quickly enough to the thermistor, causing a slight delay in the heat flux which then causes the PETG to rapidly heat due to the rapid decrease in flow. The result is over-extrusion and stringing.

Conversely, the rapid acceleration after a corner, overhang or other feature is past causes the PETG to rapidly cool when the flow rate increases, causing low viscosity that in turn causes the nozzle to drag away part of the just-printed filament.

There are a few solutions, one of which is using a nozzle with higher usable volume. The extra volume acts as a buffer, helping to decrease the effects of rapid temperature changes.

Another solution, which is what I'm proposing, is to utilize a nozzle heater pre-act, which is basically a predictive temperature adjustment made slightly in advance of a known time. In the case of PETG printing, it means initiating the nozzle heater power decrease just before reaching the point of slowdown, then increasing nozzle heater power just before resuming normal print speed. Such a pre-act would serve to moderate the temperature of the PETG in the nozzle, thereby eliminating the adverse effect on viscosity.

This pre-act could be implemented in software or firmware, or a combination of both, and should be adjustable by the end user. The nature of slicing and printing means that speed changes are known well ahead of time and can be easily accommodated.

This would be good for any highly temperature-sensitive material, not just PETG.

thepkc, 2024-08-17:
Amazing idea. This confirms all of my testing on my x1C too.

thepkc, 2024-08-17:
For reference, this apparently has been implemented as a slicer addon: https://github.com/sb53systems/G-Code-Flow-Temperature-Controller

tsmith35, 2024-08-17:
> Amazing idea. This confirms all of my testing on my x1C too.

I thought it was a good idea, but the feedback I got was to use a CHT nozzle or print slower, both workarounds. I don't have the time or skill to pursue a lookahead based PID solution, so I just left it for someone else to use. C'est la vie.