bambulab/BambuStudio #2794: Printing in mid-air

jwr, 2023-10-26:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 13.6

### Printer

Bambu X1C

### How to reproduce

1. Slice the supplied file with "0.20mm Standard @BBL X1C" preset.
2. Go to layer 11.
3. Observe how the generated G-code prints mid-air, first trying to print the inside of the hole, and then filling in the outside.

Increasing wall loops to 3 makes this problem disappear in this particular case.

### Actual results

[image]

### Expected results

I expected the slicer to print the hole from the outside in, so as not to try to print in mid-air, which is guaranteed to fail.

The results with wall loops set to 3 are what I would expect.

### Project file & Debug log uploads

[Slicer issue v3.3mf.zip](https://github.com/bambulab/BambuStudio/files/13181287/Slicer.issue.v3.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-10-30:
You are right,  but it a bit tricky to figure out that. i will try.

coreyward, 2023-12-19:
I have a similar issue. Bambu Studio is trying to create a feature over nothing, then connect bridges to it. This is failing when printed—the stacked layers over nothing caused 6 parts to fail in the same way.

[image]

The very next layer:
[image]

A few layers later, showing what it is building:
[image]

Final piece for reference:
[image]

---

Prusa Slicer handles this in a much safer way. This is the same layer as shown in the first screenshot, but in Prusa Slicer. You can see it does not print the inner circle:
[image]

This is the very next layer in Prusa Slicer. You can see that it bridges the full surface:
[image]

A few layers later, Prusa Slicer has created the necessary geometry on top of the bridge, but without yet creating the cylinder that appears on a higher layer. This works much better:
[image]

Here is a demo 3MF of a similar file for reference: 
[Issue Demo.3mf.zip](https://github.com/bambulab/BambuStudio/files/13719171/Issue.Demo.3mf.zip)

coreyward, 2024-01-06:
@QingZhangBambu Any luck finding time to look into this? I run into it frequently—the slicer’s proclivity to draw circles in the center to do incomplete bridges rather than creating a full bridge layer and draw the circle on top creates lots of failures, so I think it's worth the time to investigate.

Here's another example of an even simpler model that can't be printed without more infill than is needed because of this behavior:
[image]

QingZhangBambu, 2024-01-09:
@coreyward the case you provide should be considered too.

QingZhangBambu, 2024-10-14:
@coreyward your issue had been fixed and please wait for next version.