bambulab/BambuStudio #2795: Upload to Makerworld ONLY option shown for disposition of sliced file

jcnrbl, 2023-10-27:
### Bambu Studio Version

1.7.7.89  1.7.7  1.7.6

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora Linux 38

### Printer

not applicable

### How to reproduce

Run appimage
Import a model
slice plate

### Actual results

Impossible to send print to printer or exposr file to SD card.

### Expected results

Should see options to print or export sliced file

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-11-01:
Hello @jcnrbl 

could you give a snapshot of your issue

jcnrbl, 2023-11-06:
This issue was due to a user error. Bambu Studio display image was not compatible with the system's display resolution IE: displayed image went beyond sright side of screen so that part of image was not displayed.

Issue closed.

jcnrbl, 2023-11-06:
Issue closed.

closed 2023-11-06 by jcnrbl