bambulab/BambuStudio #2798: Missing Elements When Slicing

BitCoinFlorida, 2023-10-27:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS X 14, WIndows 10

### Printer

X1C

### How to reproduce

1. Use any nozzle size and slice, the embedded letters will slice properly
2. Clone the item and slice again

### Actual results

The text is flush with the surface, when cloned, one of the items slices properly and the other has a number missing. This also happens with another file I created without cloning it.

I'm testing again now and I cannot get the smaller piece to do it again like shown in the photo, but you can see the larger image is doing it every time, even without cloning. The only way I could get it to work 100% of the time is to raise the text .2mm so it is sitting on the surface and not flush.

[image]

### Expected results

Text should slice normally.

### Project file & Debug log uploads

[12H2 Encoder Template.3mf.zip](https://github.com/bambulab/BambuStudio/files/13187919/12H2.Encoder.Template.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

BitCoinFlorida, 2023-10-27:
Here is another example, it's one of the files that was included in the original upload. The first image is after it is sliced, the second image is after I removed the body and then sliced it. The text was generated on a plane, so the surface is exactly the same as each letter/number. 
[image]
[image]

divingmitchy, 2023-10-28:
Yeah i had the same problem, and after trying to fix it, it just got worse:-( 
hope there is a solution asap.

BitCoinFlorida, 2023-10-29:
Someone suggested changing the part to a modifier and that worked, but not 100%. Once it's changed, it will work, then you clone it and the clone may or may not work properly. In this case, the clone did not work but all the parts were modifiers. I toggle one of the modifiers to a part and back to a modifier and the overall part works properly. Video attached has no audio: https://app.screencast.com/gveGy3a2s7Rl6

zhimin-zeng-bambulab, 2023-11-30:
[image]
@BitCoinFlorida  You can move "body2" behind "24H2 Encoder Template-3" as shown in the picture.

When there are multiple parts under an Object that overlap each other, the Parts later in the list are displayed with higher priority.

zhimin-zeng-bambulab, 2023-12-19:
@BitCoinFlorida  Did the above method solve your problem?

BitCoinFlorida, 2023-12-19:
@zhimin-zeng-bambulab thanks for responding. I didn't see your previous suggestion. Ordering didn't do it. I had to change to a modifier then it worked. When I cloned the item afterwards, it broke it again but if I toggled one of the elements to a part and then a modifier, it fixed it. Still looks like a bug. Here is a video of it: https://app.screencast.com/gveGy3a2s7Rl6?fbclid=IwAR2NyUDevAB49YHsodkdRY2DRhoQCna0v2yk2wyXIQr8jpJf_wp71_osnsE&conversation=iNCqtw0VkUGkPKi8vsRcSZ

zhimin-zeng-bambulab, 2024-02-07:
This problem is because when copying Object, the order of the parts is reset. We will fix it in the next version.

closed 2024-06-26 by zhimin-zeng-bambulab