bambulab/BambuStudio #2802: Since my P1S updated to firmware version 01.04.02.00, Bambu Studio can no longer receive video from the printer and hangs when video is requested from the status windows on the devices tab.

JavierGaldos, 2023-10-29:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Update yopur P1S to the 01.04.02.00 firmware versión
2. From Bambu studio send and object to print
3. On "device" tab
4. Click the "play" button to get video from the printer

### Actual results

No video and Bambu Studio hangs.

No problem with Bambu Handy video shows as ever.

### Expected results

Receive video to the window in Bambu Studio

### Project file & Debug log uploads

[debug_network_Sun_Oct_29_12_43_27.log](https://github.com/bambulab/BambuStudio/files/13198821/debug_network_Sun_Oct_29_12_43_27.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-10-30:
Does your so-called hang mean that the program cannot receive mouse input? Or is the video always loading? Anyway, please provide all the logs in the log directory. If the program cannot receive mouse input, please provide the DMP file according to the following wiki.

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

JavierGaldos, 2023-10-31:
Thank you very much for your time.
I thought that uninstalling and reinstalling the program might fix the
issue, but it didn't. The problem still occurs even with a fresh install.
The program cannot receive mouse input  after clicking to activate the
video.
I've attached all the log files and the DMP from the folder. You can find
those files in:
https://drive.google.com/file/d/1HWOk6IL1W1nDywO-Up3gW8JNgEeV29oF/view?usp=sharing

Javier Galdos

Quidquid latine dictum sit, altum videtur.

bambu123, 2024-02-06:
@JavierGaldos 
Sorry for late response. 
From the DMP, it seems a d9d3 compatible problem. Please refer the wiki and try config graphics card for Studio.

https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view