bambulab/BambuStudio #2803: Misplaced message when opening a A1-profile with “print by object”

coat64, 2023-10-30:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1 mini

### How to reproduce

When opening a profile/3mf for the A1-mini with “print by object” in it, the slicer gives a warning that timelapse does not work and gives the choice if you want to continue with “print by object”. That choice is given at the wrong time and is at best a message at the right corner. Of course this is meant to be a desired setting and timelapse is secondary to that.

If you choose “No” in the given example you go from a print with 1 filament change to a print with 191 changes :slight_smile:

### Actual results

If you choose “No” in the given example you go from a print with 1 filament change to a print with 191 changes.

### Expected results

No such question when opening the file. The user can be confused for no reason and pick the wrong option

### Project file & Debug log uploads

[kerstboom_A1pbo.zip](https://github.com/bambulab/BambuStudio/files/13199764/kerstboom_A1pbo.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

coat64, 2023-10-30:
Oh the same warning also pops up when closing Bambu Studio.

zhimin-zeng-bambulab, 2023-10-30:
we will fix it in the next version, thanks for report this issue!

zhimin-zeng-bambulab, 2023-11-13:
It has been fixed in v01.08.00.57, you can have a try.

closed 2023-11-16 by zhimin-zeng-bambulab