bambulab/BambuStudio #2804: Top Interface Layers "brim" around first layer of model.

Deraj103, 2023-10-30:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro, 22H2

### Additional system information

CPU: Intel(R) Core(TM) i5-7600K CPU @ 3.80GHz   3.79 GHz
Memory: 32 GBs
GPU: NVIDIA GeForce RTX 3050

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Adding a value to tree support brim width And adding top interface layers.

### Actual results

This creates a top interface layer "brim" around the points of contact of the model to the build plate. It seems to override the brims of the tree supports. Even to the point where the supports have no brim, causing them to be unstable and fall over. Increasing the tree support brim width increases the width of this top interface layer "brim".
[image]

### Expected results

If I remove the values of tree support brim width OR top interface layers, this odd brim goes away.

### Project file & Debug log uploads

[Summoned_Earth_Elemental.zip](https://github.com/bambulab/BambuStudio/files/13200490/Summoned_Earth_Elemental.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Deraj103, 2023-10-30:
Something I've noticed. Unsure if this is relevant. If I leave tree support brim width as 0, I noticed that there are a small amount of top interface layers on the first layer. This must be simply the design of the model. 
So why is the top interface layers "brim" influenced by the tree support brim width? Is it a legit feature or a bug?
I've attached a bottom view of the top interface layer "brim"
[image]

tangketan, 2023-11-06:
This bug has been fixed internally. New version will be release quite soon.

tangketan, 2023-12-21:
Offical 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan