bambulab/BambuStudio #2807: Can't paste into "Add Custom G-code"

GuzziRaz, 2023-10-31:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.1

### Additional system information

intel

### Printer

N/A

### How to reproduce

### First problem
1. Slice an object
2. Right click at a layer to insert custom G-code
3. Write a line, eg. "G90"
4. Switch to another app (eg. looking at notes)
5. Switch back to Bambu Studio for writing a second line. But the previous line is now gone!

### Second problem
1. Slice an object
2. Right click at a layer to insert custom G-code
3. Try to paste a few lines of gcode (or just a single line). You can't - nothing happens!

Orca Slicer does not have the first problem, but it shares the second problem.

### Actual results

- Switching to another app and back clears the gcode box.
- Can't paste into the box from another app [this might depend on clipboard format - I copied text from "Notes" and tried to paste it].
- You also can't move cursor around in the box. Eg. if you write
```
G90
G91
```
and then try keyboard "arrow up" to insert a line between those two lines, you can't. You also can't use the mouse to get there. Apparently you need to erase the last line in order to write anything before it.

### Expected results

- You should be able to switch to another app and back without losing content.
- You should be able to copy and/or paste.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LiZ-Li-BBL, 2023-11-09:
Hello,
We plan to improve the experience of using this feature. Also this problem https://github.com/bambulab/BambuStudio/issues/2859 will be solved.

LiZ-Li-BBL, 2024-01-16:
Has been fixed in the lastes version. https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89

closed 2024-01-16 by LiZ-Li-BBL

markrickert, 2024-09-22:
I just ran into this issue in version `1.9.7.52`