bambulab/BambuStudio #2808: Segfault when logging in

klsgh, 2023-10-31:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

openSUSE Leap 15.3

### Printer

Bambu Lab X1C

### How to reproduce

1. Launch Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImag
2. Install the network module
3. Click on "Login/Register"
4. Accept the cookies prompt
5. Enter email address and password
6. Check "I agree to Terms of Use and Privacy Policy
7. Click on "Sign In"
8. Nothing happens, so click on "Sign In" again - to no avail
9. Click on "Create Your Account"
10. In the "Register" dialog enter the same email address as in step 5
11. Click "Send verification code"
12. Gets "Invalid or already taken"
13. Close the "Register" dialog
14. Click "Login/Register" again and get the "Continue to Bambu Studio" dialog (very irritating)
15. Click "Continue to Bambu Studio"
16. Segmentation fault (core dumped)

### Actual results

Segmentation fault (core dumped)

### Expected results

Bambu Studio should keep on running

### Project file & Debug log uploads

[debug_Tue_Oct_31_17_00_54_5493.log.0.gz](https://github.com/bambulab/BambuStudio/files/13218690/debug_Tue_Oct_31_17_00_54_5493.log.0.gz)

There are no data files involved, the crash happens before anything can be done.

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-11-09:
Hello @klsgh 

we have just released v1.8.0 public beta version
and we have fixed the login related issue on linux

could you have a try to verify?
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

klsgh, 2023-11-09:
Hi, that's great! I'll try it as soon as I get home from my trip ending Nov 16.Klaus

https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

klsgh, 2023-11-17:
I just started version 1.8.0.57, installed the network plugin, logged in and after showing that I was actually logged in for a second or so, the app crashed:

`(bambu-studio:27916): Gtk-CRITICAL **: 11:48:19.555: gtk_widget_set_size_request: assertion 'width >= -1' failed
1. lo    : 127.0.0.1
2. eth0  : 192.168.1.33

No such file or directory

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.769: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.769: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.770: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.770: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.774: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.774: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.774: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.774: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.776: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.776: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.778: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.778: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:27916): GLib-GObject-WARNING **: 11:48:21.778: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:27916): Gtk-CRITICAL **: 11:48:21.778: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault (core dumped)`

lanewei120, 2023-11-17:
it seems the crash only happens on OpenSUSE now

we have tried ubuntu/fedora/archlinux
and works well here

it seems we need to install an OpenSUSE also

klsgh, 2023-11-17:
I would very much appreciate that.
If there's anything else I can test, please let me know.

klsgh, 2023-12-11:
Just tried BambuStudio_linux_ubuntu_v01.08.01.57.AppImage.
Still crashes on openSUSE 15.3.

lanewei120, 2023-12-15:
@klsgh 

please try this solution currently
https://github.com/bambulab/BambuStudio/issues/2663#issuecomment-1857870433

lanewei120, 2023-12-29:
@klsgh could you try https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage?

we have fixed this issue

klsgh, 2023-12-29:
@lanewei120 Thanks, it works!