bambulab/BambuStudio #2810: Possible bug - larger than expected gap to print plate.

rsilvers129, 2023-10-31:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10.

### Printer

Bambu X1 and also A1

### How to reproduce

Print my 3MF, kill power after print starts. Measure gap.

### Actual results

My friend ruined his print head of a new A1 when it printed a giant blob. I looked at his printer and saw that it had a larger than 0.2mm gap on the first layer. Strangely, this happened to me today for the first time ever - I had a massive blob. Then again on another one of my X1, I saw the first layer not sticking. I went to the printer and killed the power so it would stop in place. I used sheets of paper to see how big the gap was. It was 0.6mm - which is too much. I think there might be a bug in the latest slicer that makes a larger than normal gap. I saw someone else say they got their first ever blob in 1000 hours of printing yesterday, so I am concerned there is a widespread problem.

### Expected results

Should have 0.2mm gap.

### Project file & Debug log uploads

[image]
[0.6mmgap.zip](https://github.com/bambulab/BambuStudio/files/13220231/0.6mmgap.zip)
[debug_Tue_Oct_31_13_55_15_17096.log.zip](https://github.com/bambulab/BambuStudio/files/13220249/debug_Tue_Oct_31_13_55_15_17096.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-11-01:
it seems that you didn't put the plate on the heat bed.

rsilvers129, 2023-11-01:
[image]

It has a plate.