bambulab/BambuStudio #2811: Support Critical Regions Only with Raft

Jester7896, 2023-10-31:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 Series

### How to reproduce

Enable Multiple Raft Layers
Enable Tree Support (Auto)
Check "Support Critical Regions Only"

### Actual results

[image]
[image]

### Expected results

It should slice

### Project file & Debug log uploads

[Blade 1.zip](https://github.com/bambulab/BambuStudio/files/13220736/Blade.1.zip)
[debug_network_Tue_Oct_31_15_31_05.log](https://github.com/bambulab/BambuStudio/files/13220747/debug_network_Tue_Oct_31_15_31_05.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2023-11-03:
We have fixed the bug internally. Will release new version quite soon.

tangketan, 2023-12-21:
Offical 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan