bambulab/BambuStudio #2820: 导入预设会消失

ZIMU1115, 2023-11-01:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows11 22H2 专业版

### Printer

A1mini

### How to reproduce

在文件中导入耗材或工艺预设后，直接关闭studio软件，并再次打开studio软件，导入的预设直接消失，再次导入会提示“有一个同名的配置 PETG ECO 标准。你想覆盖它吗？”选择“是”或“全是”也不会显示出预设参数

### Actual results

[image]
[image]
[image]

### Expected results

希望可以修复这个问题

### Project file & Debug log uploads

[天瑞PETG ECO.json](https://github.com/bambulab/BambuStudio/files/13230036/PETG.ECO.json)
[天瑞PETG 0.2层高标准设置.json](https://github.com/bambulab/BambuStudio/files/13230037/PETG.0.2.json)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bambu123, 2023-11-03:
可能与当前打印机不兼容

ZIMU1115, 2023-11-03:
> 可能与当前打印机不兼容

那不兼容也不应该直接不显示吧 至少显示不符合我当前打印机的参数 让我自行修改

bambu123, 2023-11-03:
> > 可能与当前打印机不兼容
> 
> 那不兼容也不应该直接不显示吧 至少显示不符合我当前打印机的参数 让我自行修改

不好意思。每个预设是一个整体，当他创建出来就已经与某些打印机绑定了。导入预设如果想用于其他的打印机，存在风险。
在导入之前，你是否知道导入的预设是基于哪个打印机的？如果不是针对你的打印机的，最好从原作者那边重新索要一份。