bambulab/BambuStudio #2822: Camera stream stops after a minute or two windows 11

RadsWa, 2023-11-02:
### Bambu Studio Version

V 1.7 onwards

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Additional system information
Nvidia Geforce RTX 3060 Ti
64GB ram 

### Printer

P1P

### How to reproduce

Print anything.

### Actual results

After approx. 1 minute it stops showing "live Feed" from the camera and shows a still picture, this is in all versions 1.7 onwards, since the upgrade to the camera feed to show outside the network. 
Am stuck using 1.62 until its fixed if I want to monitor prints from outside of the printer room without having to press camera on and off every time. 

### Expected results

Expect it to work as before and show stream on network while printing and until finished

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13234621/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

RadsWa, 2023-11-09:
FYI orcaslicer had this same issue and it was fixed in the 1.8.0 beta, maybe that can guide where the issue is for you.

bambu123, 2023-11-17:
> FYI orcaslicer had this same issue and it was fixed in the 1.8.0 beta, maybe that can guide where the issue is for you.

You can try our v1.8 beta version. [1.8.0 Public beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57)