bambulab/BambuStudio #2824: Studio home screen not showing projects in Ubuntu

Gerd-St, 2023-11-02:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04.3 with latest updates

### Additional system information

CPU: AMD® Ryzen 7 5700g
Memory: 16gb
Graphics: NVIDIA GeForce RTX 3060/PCIe/SSE2

### Printer

P1S

### How to reproduce

1. Just start the AppImage, no project file required

### Actual results

The home screen will show up empty with white background
When moving the mouse around you will find clickable items, just not visible
All other functions of Studio seem to work as expected

### Expected results

After AppImage startup, the Studio home screen should display online project samples.

[image]
[image]

### Project file & Debug log uploads

[debug_Thu_Nov_02_08_45_18_6022.log.zip](https://github.com/bambulab/BambuStudio/files/13235947/debug_Thu_Nov_02_08_45_18_6022.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

NXT, 2023-11-02:
Hi, you should install libnvidia-egl-gbm1. That should work with nvidia installed stuff. It worked for me.

lanewei120, 2023-11-03:
@Gerd-St 
if the above way still doesn't work,
Could you also have a try following the methods used in https://github.com/bambulab/BambuStudio/issues/1828

Gerd-St, 2023-11-03:
Installing libnvidia-egl-gbm1 didn't help.
But when running WEBKIT_DISABLE_COMPOSITING_MODE=1 ./Bambu_Studio_Linux_*.AppImage from terminal the home screen showed up as should.
Can I include this setting into the desktop file, too, so there is no need for starting via terminal?

bambu123, 2023-12-25:
We'll add WEBKIT_DISABLE_COMPOSITING_MODE to code.

lanewei120, 2024-05-14:
already added it in 1.9.1

closed 2024-05-14 by lanewei120