bambulab/BambuStudio #2826: Crash on Bambu Studio 1.7.7.89

ireneusbs, 2023-11-02:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11 Pro

### Additional system information

Prozessor: Intel(R) Xeon(R) CPU E5-1660 v3 @ 3.00GHz, 3001 MHz, 8 Kern(e), 16 logische(r) Prozessor(en)
RAM: 32,0 GB
Grafik: Intel ARC A770 16GB

### Printer

BambuLab P1P

### How to reproduce

After Sending Job to Cloud (finished) the Studio crashed and closed.
Not allways, but sometimes (till now 3 times in 1 Week).

### Actual results

BambuStudio closed

### Expected results

Nothing, Job is printing on the P1P

### Project file & Debug log uploads

[bambu-studio.exe.2788.zip](https://github.com/bambulab/BambuStudio/files/13243516/bambu-studio.exe.2788.zip)
[CrystalDragon_NewLegs_ShortTail_Rectangle.zip](https://github.com/bambulab/BambuStudio/files/13243525/CrystalDragon_NewLegs_ShortTail_Rectangle.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lhong820, 2023-11-08:
This issue exists for me also.

bambu123, 2023-11-14:
@ireneusbs To analyze DMP files, we need you to provide the following DLLs on your system:

C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.22621.2215_none_270d72677387dfc3\comctl32.dll
C:\Windows\System32\ntdll.dll
C:\Windows\System32\kernel32.dll
C:\Windows\System32\KERNELBASE.dll
C:\Windows\System32\user32.dll
C:\Windows\System32\msvcp140.dll
C:\Windows\System32\ucrtbase.dll
C:\Windows\System32\win32u.dll

bambu123, 2023-11-14:
#2723

bambu123, 2023-12-25:
@ireneusbs We have released version v1.8, please try using the new version to see if the issue exists.