bambulab/BambuStudio #2827: Steam OS(Arch Linux) support

hogum, 2023-11-02:
Hopefully support for steam os will be added so we can do slicing on the steam deck :)

Schokobecher, 2023-12-07:
Without owning a SteamOS device:
What is the issue with the AppImage? Or even one of the available AUR packages?

lanewei120, 2023-12-18:
how about run the appImage of ubuntu on steam OS?
it can be run well on arch linux before after install some libraries such as  webkit2gtk

Schokobecher, 2023-12-18:
@lanewei120 the point of an AppImage is to bundle all required libs, that would include _webkit2gtk_
Some GNOME-based distros come with it pre-installed, but not that shouldn't be what is relied on, otherwise providing a DEB/RPM etc. would make more sense as it can load the required dependencies by itself.

hogum, 2023-12-18:
> @lanewei120 the point of an AppImage is to bundle all required libs, that would include _webkit2gtk_ Some GNOME-based distros come with it pre-installed, but not that shouldn't be what is relied on, [...]

Agree. On steam os, using pacman to install software is very troublesome. If the software package requires dependencies, it is easier to use Flatpak.

closed 2024-02-14 by hogum