bambulab/BambuStudio #2828: "Failed to add custom font"

moritz-t-w, 2023-11-02:
### Bambu Studio Version

bambustudio-01.07.07.89-1

### Where is the application from?

Built from source by yourself or others

### OS version

Linux pc 6.5.9-arch2-1 #1 SMP PREEMPT_DYNAMIC x86_64 GNU/Linux

### Additional system information

n/a

### Printer

n/a

### How to reproduce

1. Install bambustudio package with pacman
2. Launch Program

### Actual results

```
[2023-11-02 22:46:01.371536] [0x00007fea12ac2480] [trace]   Initializing StaticPrintConfigs
22:46:01: Error: Failed to add custom font "/home/moti/.cache/yay/bambustudio/src/install_dir/share/BambuStudio/fonts/HarmonyOS_Sans_SC_Bold.ttf".
add font of HarmonyOS_Sans_SC_Bold returns 0
22:46:01: Error: Failed to add custom font "/home/moti/.cache/yay/bambustudio/src/install_dir/share/BambuStudio/fonts/HarmonyOS_Sans_SC_Regular.ttf".
add font of HarmonyOS_Sans_SC_Regular returns 0
```

### Expected results

Bambu Labs starting ig

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

boredomwontgetus, 2023-11-09:
Same for me with the AUR package:

`
❯ yay -Q bambustudio
bambustudio 01.07.07.89-1

❯ bambu-studio
[2023-11-09 19:00:09.170999] [0x00007fadff279480] [trace]   Initializing StaticPrintConfigs
19:00:09: Error: Failed to add custom font "/home/tom/.cache/yay/bambustudio/src/install_dir/share/BambuStudio/fonts/HarmonyOS_Sans_SC_Bold.ttf".
add font of HarmonyOS_Sans_SC_Bold returns 0
19:00:09: Error: Failed to add custom font "/home/tom/.cache/yay/bambustudio/src/install_dir/share/BambuStudio/fonts/HarmonyOS_Sans_SC_Regular.ttf".
add font of HarmonyOS_Sans_SC_Regular returns 0
`

moritz-t-w, 2023-11-10:
yes I'm also using the AUR pkg

maks, 2023-12-11:
Unfortunately same here. I this is because the src is not uncompressed in that location:
```
.cache/yay/bambustudio/src/': No such file or directory
```

flying-x, 2024-01-04:
I had the same problem. After switching to bambustudio-git, it seems to be OK now.

```
[2024-01-03 19:38:03.034532] [0x00007f519f498480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
```

lanewei120, 2024-03-01:
> Unfortunately same here. I this is because the src is not uncompressed in that location:
> 
> ```
> .cache/yay/bambustudio/src/': No such file or directory
> ```

yes, exactly

@flying-x 
so what is the original git used?

flying-x, 2024-03-01:
Here is the package of bambustudio and bambustudio-git:

https://aur.archlinux.org/packages/bambustudio-git

https://aur.archlinux.org/packages/bambustudio

You can git clone their build files:

[image]

My build dir looks like:

[image]

The PKGBUILD file is a script for building. It contains the prepare, build and package steps:

[image]

I think the maintainer there needs some help for both packages since both of them have different problems running on arch. Right now, I can only use the binary package or the flatpak packages.

flying-x, 2024-03-01:
comment from the maintainer:
[image]

moritz-t-w, 2024-04-05:
@RPINerd:
> Totally naive question, is there just a directory somewhere in the install directory to 'gunzip'? Or is this a deeper problem with the actual building process?

Totally naive answer: you could probably even install it the normal way, or by copying it into the fonts folder, which is `/usr/share/fonts/` for me