bambulab/BambuStudio #2831: System Presets override (as opposed to overwrite)

GuzziRaz, 2023-11-03:
### Is your feature request related to a problem? Please describe.
System Presets are not *overwriteable*, which is indeed a safe design choice. But this causes custom settings, such as (but not limited to) filament price, build plate temperatures or ABS fan speeds hard to maintain using User Profiles - because they all inherit the parent system preset: As soon as I change nozzle, all prices are reverted. As soon as I create a new calibration profile, everything is reverted to system defaults I need to manually copy any custom settings all the time.

### Describe the solution you'd like
System Settings should be ***overrideable*** as opposed to ***overwritable***. What I mean is this:

1. I should be able to (for example) select `Bambu PLA Basic @BBL X1C` filament system preset, change Price to a figure that is applicable to me (eg. some european price including VAT), and save it *without creating a user preset*. It would *appear* as if I overwrote the system preset, but that's not quite happening under the hood:
2. The actual result would be a "system preset override". It will be a level between the system presets and the user presets. There will be no new name involved (at least not visible to the user).
3. After changing this single instance of Price, other system presets as well as new or already existing user presets *that inherits* that system profile (such as a flow rate calibrated user preset, or the very same filament but for a 0.2 nozzle) would reflect the new price. When the price changes in the future, I can update it in that one single place - and it will again be inherited down properly.
4. As long as eg. "price" is overridden, it would show the "revert" icon ↻ beside it when viewing the system preset - and clicking it would remove that particular override. A revertion will, obviously, be inherited by all and any children.

### Describe alternatives you've considered
A workaround is to manually edit system presets json files using a text editor. The changes will then be inherited correctly. This works just fine but is indeed a hack and may break or get lost with software or OTA updates. Also, I can't revert changes but need to use backup copies.

### Additional context
This is not just about filament price but just about ANY setting, such as temperatures or max volumetric speed. This whole idea should also not only affect filaments but also printers (such as `Bambu Lab X1 Carbon 0.4 nozzle`) or processes (such as `0.20mm Standard @BBL X1C`).

You could likely use the existing inheriting scheme for implementing this pretty trivially.

OzDruiD, 2023-12-04:
I agree, the system presets, possibly some of the pricing and such, should be modifiable, as well as things like temps etc
Not always do manuafacture recommended temps work best, altitude, moisture levels, ambient temp etc.  Once your printer is dialed in, it might alweays be using different settings - and yess Custom filaments could duplicate Systems and adjust those - but ... why.....

I am guesing support reasons.. but 100% - there should be a proteted layer, a user layer,, and amiddle ground for pre-defined, known vendor or "system" presets.

tsmith35, 2024-01-22:
Yes, a means of providing users with a persistent set of overrides would solve many issues. I'm tired of the constant battle to get things improved.

For example, BBL insists that X1E users have enough experience to control the chamber fan, but X1C/P1S users do not ... what? While I strongly disagree with that assertion, trying to sell my point is like trying to change the weather. BBL doesn't want to listen.

So how about providing individual users with a means to customize settings to our hearts content without anyone at BBL having to lift a finger? I may be too dumb to control my own chamber fan, but at least I won't have to resort to using Grep to modify the BBL profiles every time I upgrade to a new software version. Yeah, I get it... BBL doesn't want new users playing in the street, but for God's sake, many users of BBL products run huge print farms, program and operate industrial automation, create massive 3D structures with all kinds of complex analyses, program and configure robots, and who knows what else. These people may be new to BBL, but they certainly aren't technologically challenged.

BBL opened the door for massive advancement in FDM. Why treat us like we're $99 Ender purchasers? Let us control the things we need to control. If it doesn't work, we're smart enough to figure that out. We may not design and build 3D printers for a living, but I'll bet you we can figure out how to get the most out of those printers. I'm not an X1E owner, but I did sleep at a Holiday Inn Express last night... lol

User overrides. Multiple levels. Apply to profiles meeting specified criteria. Provide as many warnings as you like, but let us make the modifications that you would not. We'll get the best of both worlds; you'll get to provide a "safe" environment for the beginners, and we'll be able to set up our printers the way that works best for us. Win-Win

Cyth, 2025-12-10:
I am having this issue too. Please fix, I think this is a bug, not a feature.

tsmith35, 2026-03-01:
I gave up on trying to help Bambu Lab be more customer-friendly. They don't want to hear it, and I'm tired of trying. It's their business, and if they don't want to accept customer suggestions, then that's their right.

I have a bunch of Bambu Lab printers, but my newest printers are Snapmaker U1s. I'm happy with them so far, and I find the features more conducive to making good, quick, multicolor prints with minimal filament waste.

Use whatever workarounds you need to continue making successful prints, but you may as well give up trying to help Bambu Lab improve; they just don't want to hear it.