bambulab/BambuStudio #2833: Request

renemosner, 2023-11-03:
@lanewei120
I have a question for you. I can take the source code of the latest release v1.7.7.89, insert the Czech language I created into the code, plus edit two files so that the Czech language can be set in BambuStudio. Then compile the entire code on BambuStudio CZ and save it on my github and publish it? The point is that people in the Czech Republic would welcome Czech in order to better understand the print settings. And I would hate to violate the license, that's why I'm writing if it's possible this way. 
Thank you for your reply, best regards, Mošner

lanewei120, 2023-11-04:
Hi @renemosner 

We have added Czech language support on the master branch, which will be shown in next version v1.8

and we have refined the flow on how to contribute languages here
https://wiki.bambulab.com/en/bambu-studio/contribute-languages

you can follow the guide to update the translations

and currently you should work on the master branch, which will be published soon

any questin, please contact us

Thanks and Regards

renemosner, 2023-11-04:
@lanewei120
Thank you for answer.  
I think I have already done that and sent you the modified text.json file in cz.  But you didn't answer my question.  I can post my own release based on your source code v1.7.7.89 with edits and insert Czech then build and post on my github.  
Thank you for answer.

lanewei120, 2023-11-04:
> @lanewei120 Thank you for answer. I think I have already done that and sent you the modified text.json file in cz. But you didn't answer my question. I can post my own release based on your source code v1.7.7.89 with edits and insert Czech then build and post on my github. Thank you for answer.

yes, you can do that
as this is an open source project:)

renemosner, 2023-11-09:
@lanewei120
Thank you very much for including Czech in BambuStudio.

closed 2023-11-10 by renemosner