bambulab/BambuStudio #2834: BambuStudio saving corrupted 3mf files that can't be opened in diffrent slicers.

adamsatyr, 2023-11-03:
### Bambu Studio Version

01.07.07.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Unkown

### How to reproduce

Try reading 3mf files saved by BambuLab on PrusaSlicer.

### Actual results

[image]

### Expected results

PrusaSlicer should be able to load those files, even just as models without settings.

I found this on 2 separate occasions (one got fixed with simple STL):
(latest) https://www.printables.com/model/627735-bonsai-tree-jewelry-display/

### Project file & Debug log uploads

Files from the link before, unable to open them in PrusaSlicer.
[Uploading bonsai-tree-jewelry-display-model_files.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

adamsatyr, 2023-11-03:
Comparing to PrusaSlicer 3mf file, it doesn't use multi .model files linked to one 3dmodel.model file, but single 3dmodel.model file that contains everything.

adamsatyr, 2023-11-03:
https://github.com/bambulab/BambuStudio/issues/169

Auto duplicate myself, welcome to self-service era.

But it seems that people are not aware of the export as Generic 3MF

MattAllison, 2023-11-18:
I believe this is actually a bug.

Previously (looking at files I created in early August), Export Generic 3mf would create a file that could be read by tools like Meshmixer and 3d Builder.

The current version of Studio creates files that cannot be read by those tools.

Something changed.

MattAllison, 2023-11-18:
When opening one of the exported 3MF files in Orca Slicer 1.7, it reports the list of unsupported keys within the file.  I'm wondering if any of these are really important in a "generic" 3mf file.

> The 3mf's version 1.7.7.89 is newer than Orca Slicer's version 1.7.4.52, Found following keys unrecognized:
> -activate_air_filtration;
> -best_object_pos;
> -chamber_temperatures;
> -complete_print_exhaust_fan_speed;
> -during_print_exhaust_fan_speed;
> -extruder_clearance_max_radius;
> -extruder_type;
> -initial_layer_flow_ratio;
> -printer_structure;
> -sparse_infill_anchor;
> -sparse_infill_anchor_max;
> -support_air_filtration;
> -support_chamber_temp_control;
> -thumbnail_size;
> -time_lapse_gcode;
> -top_one_wall_type;
>

DanBao-Bambu, 2023-11-20:
The three 3mf files (Trunk, Pot, and Leaves) in the link you provided cannot be directly opened in PrusaSlicer. But I can open it using the latest 1.8 Beta Bambu Studio and export a generic 3mf file that can be opened by PrusaSlicer. Regarding the function of exporting general 3mf files, Bambu Studio version 01.07.07.89 has indeed been modified, but the latest 1.8 Beta Bambu Studio has been fixed. You can try using the latest Studio. If you have any further questions, please reply to me.