bambulab/BambuStudio #2836: Ability to save flow dynamics calibration that is done during the the beginning of the printing phase

kr320, 2023-11-03:
After selecting Print Plate, if Flow Dynamics Calibration is checked, an additional option could pop up to save the calibrated result. Another option could be to view the calibrated result and choose to save it if you'd like to.

If it is a multicolor print it would have to state that it can only do the calibration/save for the first filament.

With this feature you wouldn't need to go into the calibration tab in order to calibrate every filament, the results will be saved and/or viewed at that time.

GuzziRaz, 2023-11-05:
Agreed. I would also like an option to skip the flow calibration automagically if one was already stored. The current option for "Flow Calibration" when sending a print would be "Flow Calibration if needed".

mario54bro, 2023-11-10:
I would like to expand on this a bit. I think we should be able to save the Flow Ratio to the AMS slot. We already have the K value per AMS slot, why not add an option to adjust the Flow Ratio per slot. The X1C can adjust Flow Ratio on the fly as it has the option to calibrate it at the start of the print. 

This would be very helpful for users that have multiple printers that are printing the same object and would like to adjust the flow ratio for different spools of filament. Currently, we have to go back to the slicer and adjust the flow ratio, re-slice it with the corrected flow ratio and then send the new file to the printer to start printing.