bambulab/BambuStudio #2839: Camera not working - Windows Media Player wants to be enabled, but it is

Osmodia666, 2023-11-04:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 22H2

### Printer

X1 Carbon

### How to reproduce

Open App. go to Camera Tab

### Actual results

Error Message " Windows Media Player is required for this Task!...."

### Expected results

Camera Feed should start

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13257485/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Osmodia666, 2023-11-04:
[image]

bambu123, 2023-11-06:
[Microsoft-Windows-MediaPlayer-Package~31bf3856ad364e35~amd64~~.zip](https://github.com/bambulab/BambuStudio/files/13263733/Microsoft-Windows-MediaPlayer-Package.31bf3856ad364e35.amd64.zip)

The user's system version of MediaPlayer may not work, try this offline package. First delete the system's MediaPlayer, and then install it with the following command.

Dism.exe /Online /Add-Package /PackagePath:C:\Microsoft-Windows-MediaPlayer-Package~31bf3856ad364e35~amd64~~.cab

s3ppel, 2023-11-06:
Step 1: Type control panel in the Search bar, and then click the best match one to open it.

Step 2: Set View by: Small icons and then click Programs and Features.

Step 3: Click Turn Windows features on or off in the left panel. In the newly pop-out window, expand Media Features and check the box next to Windows Media Player. Read the message, click Yes and OK.

Step 4: Restart your computer

Osmodia666, 2023-11-06:
> Step 1: Type control panel in the Search bar, and then click the best match one to open it.
>
> Step 2: Set View by: Small icons and then click Programs and Features.
>
> Step 3: Click Turn Windows [...]

Thank you, this worked fine.

closed 2023-11-06 by Osmodia666

Majestic7979, 2023-12-11:
BambuLab, this is garbage. Why don't you fix your software instead of asking people to rely on LEGACY software? You have a state-of-the-art modern printer and you're relying on a no-longer-supported feature of WINDOWS 11!? HELLO! WMP is a relic from a bygone WinXP era. STOP ASKING USERS TO DO SILLY WORKAROUNDS FOR OUTDATED SOFTWARE.

XeviaN360, 2024-02-10:
> BambuLab, this is garbage. Why don't you fix your software instead of asking people to rely on LEGACY software? You have a state-of-the-art modern printer and you're relying on a no-longer-supported [...]

This.

tordsta, 2024-05-02:
> Step 1: Type control panel in the Search bar, and then click the best match one to open it.
>
> Step 2: Set View by: Small icons and then click Programs and Features.
>
> Step 3: Click Turn Windows [...]

This worked for me as well, thanks!

julie777, 2025-07-25:
I am using Windows 10 22H2 and there is no Windows Media Player. There is only Media features which is turned on.

[image]