bambulab/BambuStudio #2840: docker build fails on undefined reference to `SSL_get1_peer_certificate'

Achllle, 2023-11-04:
### Bambu Studio Version

master

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 22.04

### Additional system information

Docker version 20.10.16, build aa7e414

### Printer

P1P

### How to reproduce

1. clone latest master branch (18d6b60b89be37022e8d6c94e54e687aa767e489)
2. bash DockerBuild.sh

### Actual results

build step fails with:
```
#14 774.9 [100%] Linking CXX executable bambu-studio
#14 786.4 /usr/bin/ld: ../../deps/build/destdir/usr/local/lib/libcurl.a(openssl.c.o): in function `servercert':
#14 786.4 openssl.c:(.text+0x2320): undefined reference to `SSL_get1_peer_certificate'
#14 786.4 /usr/bin/ld: openssl.c:(.text+0x2f46): undefined reference to `SSL_get1_peer_certificate'
#14 786.4 /usr/bin/ld: openssl.c:(.text+0x3187): undefined reference to `EVP_PKEY_get_id'
#14 786.4 /usr/bin/ld: ../../deps/build/destdir/usr/local/lib/libcurl.a(openssl.c.o): in function `ossl_connect_step1':
#14 786.4 openssl.c:(.text+0x4b4d): undefined reference to `EVP_PKEY_get_id'
#14 786.4 /usr/bin/ld: openssl.c:(.text+0x4c78): undefined reference to `SSL_CTX_load_verify_file'
#14 786.4 /usr/bin/ld: openssl.c:(.text+0x4cc4): undefined reference to `SSL_CTX_load_verify_dir'
#14 786.4 collect2: error: ld returned 1 exit status
#14 786.4 make[3]: *** [src/CMakeFiles/BambuStudio.dir/build.make:188: src/bambu-studio] Error 1
#14 786.4 make[2]: *** [CMakeFiles/Makefile2:596: src/CMakeFiles/BambuStudio.dir/all] Error 2
#14 786.4 make[1]: *** [CMakeFiles/Makefile2:603: src/CMakeFiles/BambuStudio.dir/rule] Error 2
#14 786.4 make: *** [Makefile:236: BambuStudio] Error 2
------
executor failed running [/bin/sh -c ./BuildLinux.sh -s]: exit code: 2
```

### Expected results

docker build completes successfully

### Project file & Debug log uploads

log included in actual results

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-11-06:
hello @Achllle 

please pick this patch https://github.com/bambulab/BambuStudio/commit/cd5216c85567c6144627a6cf443670869e5fc88d for your building on master

we will sync the codes to master soon

Achllle, 2023-11-06:
Confirming that commit fixed the issue. Closing

closed 2023-11-06 by Achllle