bambulab/BambuStudio #2842: 打开任何3mf文件，报错0x00007ffa1b633168

hansj1984, 2023-11-05:
### Bambu Studio Version

v01.07.07.89

### Where is the application from?

Bambu Lab github releases

### OS version

win10 22H2

### Printer

P1S

### How to reproduce

使用Bambu_Studio打开任何3mf文件都报错，报错提示   0x00007ffa1b633168

[image]

### Actual results

报错

### Expected results

报错

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-11-06:
@hansj1984 麻烦参考下面的 wiki 提供 Studio 日志给我们。特别不要遗漏 DMP 文件。

https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

bambu123, 2023-12-22:
@hansj1984 请问你的问题解决了吗？