bambulab/BambuStudio #2846: Arc Fitting Only Works on External Perimeters

CCS86, 2023-11-05:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Slice a model of a cylindrical tube and look at the g-code output for external and internal perimeters.

### Actual results

In trying to diagnose why a mostly cylindrical print sounded smooth on some perimeter loops, and sounded very grunty on others, I noticed the arc fitting is only being applied to external perimeters, and even perfectly circular internal perimeters are still interpolated with G1 moves. This is true on both classic and arachne engines.

### Expected results

Why not fit arcs everywhere (besides slicing time)? At least give us the option so the printer runs more smoothly and more quietly. Plus, internal perimeter artifacts will print through to external perimeters.

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/bambulab/BambuStudio/files/13255782/Cylinder.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

CCS86, 2023-12-01:
Any updates on this @QingZhangBambu ?

QingZhangBambu, 2023-12-04:
I'll check out the arc fitting process, it might be a little delayed since we're updating to the 1.8 beta version.

SaltWei, 2024-08-09:
@QingZhangBambu Bump

CCS86, 2024-10-29:
?

CCS86, 2024-12-03:
@SaltWei @QingZhangBambu