bambulab/BambuStudio #2847: [Printer Feature] Changing Nozzle Mode

MiscPrinter, 2023-11-05:
**Is your feature request related to a problem? Please describe.**
When changing a nozzle, one has to turn off the machine, change the nozzle, turn the machine back on. This is not ideal. You loose the bed leveling when power cycling the machine along with the time to reboot and any pre-heating of the bed. 

**Describe the solution you'd like**
Add a button on screen to tell the printer you are changing the nozzle; don't freak out when it looses the temperature reading. This button would turn off the heater/fan and wait until the user swaps the nozzle. Then the user would hit the button again to tell the printer, the nozzle has been changed; resume normal operation.

mzient, 2023-11-25:
Have you considered using a third party hotend with replaceable nozzle? Replacing the entire hotend "the bambu way" is not viable mid-print anyway, because the assembly is not self-centering and the print will be misalinged. Having said that, even for the third party-nozzles, recalibrating the nozzle height is a must.

MiscPrinter, 2023-11-25:
I had and tried after market hotends. Their worse. The set screws that hold the throat to the heatsink do not grab the throat enough to torque the nozzle. You have to use a wrench. The CHT variants also cause more jams. 

The point is not to change nozzle size mid print. It changing in between prints when switching from, say,  a .4mm to .2mm for the next print. Having to power off, switch nozzle, then power on the machine takes more time than leaving the machine on while putting it into a special "nozzle change mode" which disabled the thermal runaway, turn off the heater, and turn off the fan.

GuzziRaz, 2023-11-26:
> When changing a nozzle, one has to turn off the machine, change the nozzle, turn the machine back on. This is not ideal. You loose the bed leveling when power cycling the machine along with the time to reboot and any pre-heating of the bed.

I second your request but just out of curiosity: What happens (using current firmware) if you just leave power on (but obviously all temps and fans off and possibly even in standby mode) and change it? You mention thermal runaway protection, does that kick in so you need to power off anyway?