bambulab/BambuStudio #2850: Latest Bambu Studio version crashed on M2 MacBook Pro

T3chC0wb0y, 2023-11-07:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 14.1

### Additional system information

Apple M2 Pro
32GB RAM

### Printer

N/A

### How to reproduce

Attempt to start Bambu Studio

### Actual results

Bambu Studio Crashed on Startup

### Expected results

The app should open as expected

### Project file & Debug log uploads

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [2861]
Path:                  /Volumes/*/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.07.07.89 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-11-06 21:49:37.8759 -0600
OS Version:            macOS 14.1 (23B74)
Report Version:        12
Anonymous UUID:        219BCD43-71EF-9BEA-A183-1ACA442B22D6

Time Awake Since Boot: 820 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000005a0
Exception Codes:       0x0000000000000001, 0x00000000000005a0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2861]

VM Region Info: 0x5a0 is not in any region.  Bytes before following region: 4336253536
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102760000-105ebc000    [ 55.4M] r-x/r-x SM=COW  ...S/BambuStudio
[... 513 lines cut ...]
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci1b73,1100, USB eXtensible Host Controller, Thunderbolt@6,0,0
PCI Card: pci1b73,1100, USB eXtensible Host Controller, Thunderbolt@5,0,0
PCI Card: pci1b21,1242, USB eXtensible Host Controller, Thunderbolt@4,0,0
PCI Card: ethernet, Ethernet Controller, Thunderbolt@3,0,0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB30Bus
USB Device: CalDigit Thunderbolt 3 Audio
USB Device: USB30Bus
USB Device: Card Reader
USB Device: Razer Kiyo Pro
USB Device: TONOR TC-777 Audio Device
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: TS3 Plus, CalDigit, Inc., 1, 44.1
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-11-08:
Thanks for informing, it will be fixed in the next version.

bambu123, 2023-11-13:
Please try the last beta version.
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

Please note that the Beta version uses a different user configuration directory, so it will not affect each other with the previous official version. However, you need to redo the general first boot.

bambu123, 2023-11-13:
> > Please try the last beta version. https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57
> > Please note that the Beta version uses a different user configuration directory, so it will not affect each other with the previous official version. However, you need to redo the general first boot.
> 
> Neither 1.7.7.89 or 1.08.00.57 will open on my M2 Mac.

Is there a Crash report? If so, send us the crash report. If not, please provide application logs.

bambu123, 2023-11-14:
> > > > Please try the last beta version. https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57
> > > > Please note that the Beta version uses a different user configuration directory, so it will not affect each other with the previous official version. However, you need to redo the general first boot.
> > > 
> > > 
> > > Neither 1.7.7.89 or 1.08.00.57 will open on my M2 Mac.
> > 
> > 
> > Is there a Crash report? If so, send us the crash report. If not, please provide application logs.
> 
> Initially there was a crash report, and I sent that. However, now it doesn't even provide a crash report, it just immediately closes. Where are application logs stored?

Here: 
/Users/<user>/Library/Application Support/BambuStudioBeta/

T3chC0wb0y, 2023-11-14:
I was able to get 1.7.7.89 to open by turning off another program that was controlling my sound settings.  It was a an equalizer program.  Not sure why that was causing a conflict, but it was.  If the BambuStudio app is closed, I still cannot launch in context from makerworld.  I have to launch the app first then send to Bambu Studio.

bambu123, 2023-11-15:
> I was able to get 1.7.7.89 to open by turning off another program that was controlling my sound settings. It was a an equalizer program. Not sure why that was causing a conflict, but it was. If the [...]

Besides the Crash report you gave at the beginning, are there any other crash reports?

davglass, 2023-11-15:
I get a crash randomly when launching the app. Sometimes multiple in a row until it finally opens up. Happens even when launching a new window (I have two printers and thats how I monitor both at once)

bambu123, 2023-11-16:
> I get a crash randomly when launching the app. Sometimes multiple in a row until it finally opens up. Happens even when launching a new window (I have two printers and thats how I monitor both at once)

Is there a Crash report? If so, send us the crash report.

T3chC0wb0y, 2023-11-16:
Here you go ....

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [10371]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.07.07.89 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-11-15 20:31:29.5817 -0600
OS Version:            macOS 14.1.1 (23B81)
Report Version:        12
Anonymous UUID:        219BCD43-71EF-9BEA-A183-1ACA442B22D6

Time Awake Since Boot: 11000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000005a0
Exception Codes:       0x0000000000000001, 0x00000000000005a0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [10371]

VM Region Info: 0x5a0 is not in any region.  Bytes before following region: 4301060704
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1005d0000-103d2c000    [ 55.4M] r-x/r-x SM=COW  ...S/BambuStudio

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x1011b1e50 Slic3r::GUI::MainFrame::update_side_preset_ui() + 60
1   BambuStudio                   	       0x10116e2d0 Slic3r::GUI::GUI_App::reload_settings() + 132
2   BambuStudio                   	       0x10119646c wxAsyncMethodCallEventFunctor<Slic3r::GUI::GUI_App::start_sync_user_preset(bool)::$_49::operator()(bool) const::'lambda'()>::Execute() + 244
3   BambuStudio                   	       0x103192290 wxEvtHandler::ProcessEventLocally(wxEvent&) + 172
4   BambuStudio                   	       0x103192060 wxEvtHandler::ProcessEvent(wxEvent&) + 56
5   BambuStudio                   	       0x103191eb0 wxEvtHandler::ProcessPendingEvents() + 236
6   BambuStudio                   	       0x1030f60dc wxAppConsoleBase::ProcessPendingEvents() + 112
7   BambuStudio                   	       0x10311ad28 wxEventLoopBase::DoYieldFor(long) + 52
8   BambuStudio                   	       0x10311ac7c wxEventLoopBase::YieldFor(long) + 120
9   BambuStudio                   	       0x1030f5ab0 wxAppConsoleBase::Yield(bool) + 104
10  BambuStudio                   	       0x102eb1474 wxWebViewWebKit::RunScriptSync(wxString const&, wxString*) const + 248
11  BambuStudio                   	       0x102eb18a8 wxWebViewWebKit::RunScript(wxString const&, wxString*) const + 84
12  BambuStudio                   	       0x102eb25a4 wxWebViewWebKit::AddScriptMessageHandler(wxString const&) + 312
13  BambuStudio                   	       0x100f7b4f8 WebView::CreateWebView(wxWindow*, wxString const&)::$_0::operator()(wxWebView*) const + 256
14  BambuStudio                   	       0x100f7b2c4 wxAsyncMethodCallEventFunctor<WebView::CreateWebView(wxWindow*, wxString const&)::$_1>::Execute() + 108
15  BambuStudio                   	       0x103192290 wxEvtHandler::ProcessEventLocally(wxEvent&) + 172
16  BambuStudio                   	       0x103192060 wxEvtHandler::ProcessEvent(wxEvent&) + 56
17  BambuStudio                   	       0x103191eb0 wxEvtHandler::ProcessPendingEvents() + 236
18  BambuStudio                   	       0x1030f60dc wxAppConsoleBase::ProcessPendingEvents() + 112
19  BambuStudio                   	       0x10311ad28 wxEventLoopBase::DoYieldFor(long) + 52
20  BambuStudio                   	       0x10311ac7c wxEventLoopBase::YieldFor(long) + 120
21  BambuStudio                   	       0x1030f5ab0 wxAppConsoleBase::Yield(bool) + 104
22  BambuStudio                   	       0x102eb1474 wxWebViewWebKit::RunScriptSync(wxString const&, wxString*) const + 248
23  BambuStudio                   	       0x102eb18a8 wxWebViewWebKit::RunScript(wxString const&, wxString*) const + 84
24  BambuStudio                   	       0x102eae3d8 wxWebView::GetUserAgent() const + 76
25  BambuStudio                   	       0x1013eba88 Slic3r::GUI::WebViewPanel::WebViewPanel(wxWindow*) + 2692
26  BambuStudio                   	       0x10119e118 Slic3r::GUI::MainFrame::init_tabpanel() + 328
27  BambuStudio                   	       0x10119cbd8 Slic3r::GUI::MainFrame::MainFrame() + 2868
28  BambuStudio                   	       0x10115c884 Slic3r::GUI::GUI_App::on_init_inner() + 5916
29  BambuStudio                   	       0x10115b11c Slic3r::GUI::GUI_App::OnInit() + 16
30  BambuStudio                   	       0x103050618 wxApp::CallOnInit() + 152
[... 480 lines cut ...]
Graphics: Apple M2 Pro, Apple M2 Pro, Built-In
Display: C49RG9x, 5120 x 1440, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Sep  1 2023 19:33:37 version 23.10.765.4.41.51.121 FWID 01-e2f09e46
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci1b73,1100, USB eXtensible Host Controller, Thunderbolt@5,0,0
PCI Card: pci1b73,1100, USB eXtensible Host Controller, Thunderbolt@6,0,0
PCI Card: ethernet, Ethernet Controller, Thunderbolt@3,0,0
PCI Card: pci1b21,1242, USB eXtensible Host Controller, Thunderbolt@4,0,0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB30Bus
USB Device: Card Reader
USB Device: Razer Kiyo Pro
USB Device: TONOR TC-777 Audio Device
USB Device: USB30Bus
USB Device: CalDigit Thunderbolt 3 Audio
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: TS3 Plus, CalDigit, Inc., 1, 44.1
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

bambu123, 2023-11-16:
I found that this issue is related to Studio's developer mode. It may be that you have turned on developer mode. You can try turning off the network to start Stduio, then turning off developer mode and turning on the network.

bambu123, 2023-11-16:
We will fix the problem ASAP.

T3chC0wb0y, 2023-11-16:
I was able to launch the app and turn off developer mode.  I then closed the application and attempted to "launch in context" a model from Makerwold.  I was able to do this successfully 3/3 times. Thanks for the quick response. I look forward to seeing this resolved in an upcoming patch.

bambu123, 2023-12-21:
Fixed in v1.8.2.

closed 2023-12-21 by bambu123