bambulab/BambuStudio #2851: Model support bug with Support Interface

mhwunder, 2023-11-07:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Mac 14.1

### Printer

P1P

### How to reproduce

Slice the model and there are floating support interfaces.  Not always but has happened.  I know this was an issue in a previous release that was fixed.

### Actual results

[image]

### Expected results

Support interface only where needed.

### Project file & Debug log uploads

[Floating Support.3mf.zip](https://github.com/bambulab/BambuStudio/files/13290344/Floating.Support.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

cRaZy92, 2023-11-08:
I have the same issue I think
[image]

With following support settings
[image]

mhwunder, 2023-11-10:
Try this test.  Uncheck "On buld plate only".  When I deslected that.  There were no more floating filament.  However, I also noted that it didn't put support interface in some places.  However, I am looking more into that to make sure it isn't just a branch needed for tree support.

[image]

tangketan, 2023-11-13:
This bug has been fixed in the latest [1.8 beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57).

tangketan, 2023-12-21:
Official 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan