bambulab/BambuStudio #2852: no connection to printer

GerhardPo, 2023-11-08:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

W11

### Additional system information

no changes in computer components, worked flawless before.

### Printer

X1CC

### How to reproduce

Try to connect printer to my account either on handy app or via studio. 

### Actual results

connecting stops at 80% and error code -1070 shows up. No changes to network components, printer is 3m away from router.
Router indicates very weak signal from printer.

### Expected results

connection to printer as before.

### Project file & Debug log uploads

no file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-11-09:
Hi. @GerhardPo 
On the X1C's screen, you can perform a network test on the printer. If the printer's network is indeed poor, try adjusting the orientation of the printer.

GerhardPo, 2023-11-09:
> Von "walterwongbbl" ***@***.***>
> An "bambulab/BambuStudio" ***@***.***>
> Cc "GerhardPo" ***@***.***>; "Mention"
> Datum 09.11.2023 12:35:47
> Betreff Re: [bambulab/BambuStudio] no connection to printer (Issue
> #2852)

Hi,

thx for your support. After logging out and factory reset of the printer 
I could finally bind the printer again. I own my printer since more than 
one year now and never had problems with connections. In the last 2 
weeks I had this problem now 2 times. No chances to any component 
neither on the printer nor on the router.
Is it possible, that the printer network pcb is defect?

Schöne Grüße

Gerhard Polic

walterwongbbl, 2023-11-10:
Hi. @GerhardPo 
Perhaps the issue is caused by the firmware upgrade we performed on the printer. I will provide this feedback to our firmware department. Thank you very much for your response. If you have any further questions, please feel free to leave us a message.

closed 2023-11-14 by walterwongbbl

GerhardPo, 2023-11-26:
Hi,

issue was gone with update, but came back with yesterday update.

Schöne Grüße

Gerhard Polic

Datum 10.11.2023 02:32:37