bambulab/BambuStudio #2859: [BETA] Arrow keys don't work in Custom G-Code textarea

Hologos, 2023-11-09:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.1.1

### Printer

Bambu Lab P1P + AMS

### How to reproduce

You reintroduced bug #1808, which had been fixed.

### Actual results

Arrow keys don't work anymore.

### Expected results

Arrow keys moving cursor.

### Project file & Debug log uploads

_no need_

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LiZ-Li-BBL, 2023-11-09:
Sorry for the inconvenience. We will fix it after this beta version.

LiZ-Li-BBL, 2024-01-16:
Has been fixed in the lastes version. https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89

closed 2024-01-16 by LiZ-Li-BBL