bambulab/BambuStudio #2860: Pause in GCode doesnt allow RESUME

keyoss, 2023-11-09:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

X1C

### How to reproduce

Slice a Project,
Select layer 2
Insert Pause
Print
(probably more than one time since the first time i used it it worked)

### Actual results

Notification appears and show option to  Pause-Continue-Stop
whatever you click you can not RESUME - only pause which will not switch to a resume state
[image]
[image]

### Expected results

Notification ALWAYS offers RESUME-Continue-Stop
even on Homescreen where you only see "II"

### Project file & Debug log uploads

[ghostbusters_high_OneBody0.1layer_v1.zip](https://github.com/bambulab/BambuStudio/files/13306955/ghostbusters_high_OneBody0.1layer_v1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

keyoss, 2023-11-09:
[image]

keyoss, 2023-11-09:
refs #2613

keyoss, 2023-12-20:
just a bump - 

Several people have issues with this! We are printing a lot of lightboxes where we need to clean the first layer by hand after its finished... 

For many people its impossible and a the loss of filament piles up actually.
Also if you run a business and have several printers its very hard to pay attention to the printer(s) which are nearly finishing the first layer. so its a pure loss and win to be there in the right minute.....

please address this issue - Prusa Slicer is working and doesnt have this issue actually with my tronxy's and other diy

MikeJ67, 2024-01-04:
Version 1.8.2.56 still has this issue. It just happened to me too.

I added a "pause" at the beginning of layer 2 by using the "+" on the layer-slide.
The print paused perfectly after the first layer was finished, and I was offered to resume printing by clicking "continue". But there is no "continue", nowhere (as described in the other comment above)

Is there any kind of workaround?

SaltWei, 2024-01-18:
> Is there any kind of workaround?

[image]

bam-winterfell, 2024-01-19:
hi@keyoss @MikeJ67 
Sorry for any inconvenience caused to you. We will fix this issue in the next version

GirlPurple, 2024-01-25:
Also seeing this issue, 3 of 4 prints have failed because of this error.

SaltWei, 2024-01-26:
> Also seeing this issue, 3 of 4 prints have failed because of this error.

[image]
Can not resume by these steps?

GirlPurple, 2024-01-28:
@SaltWei - I don't see a green arrow - the interface shows the orange pause button instead. Clicking on it does nothing. 

From the printer screen, it still shows me the Pause button instead of Resume.  I even tried pulling power on the unit in hopes that it would trigger the power outage resume, which didn't work either. The only way out of it is to cancel the job.

bam-winterfell, 2024-01-29:
hi@GirlPurple 
May I ask what is the version number of BambuStudio you are using. Additionally, when paused, the printer screen should be able to continue printing. Would you please record a video so that we can see where the problem is.

GirlPurple, 2024-01-29:
@bam-winterfell - 

My BambuStudio version = 1.8.4.51, which the "check for updates" help menu option says is the latest version. 

"Additionally, when paused, the printer screen should be able to continue printing." 
Indeed. The one time the resume function worked properly, the printer screen did offer a Resume button and it worked as expected. The other three failed times it showed options to Pause, Stop, or Close. The Pause button did nothing when pressed and the only working button was to stop/cancel the job.  

Please clarify what you would like me to record. Do you want a time lapse of the print or something else? Running the print again means committing to ~30 minutes of waiting for a full bed job to get to layer 2 and then most likely having it fail / waste filament, so I want to be sure I'm provding what you think will be most useful.

bam-winterfell, 2024-01-29:
@GirlPurple 
Sorry, I didn't consider that the conditions for reproduction were indeed quite strict.  Next time you encounter this situation again, please take a screenshot of the content displayed on the printer screen and the device page of BambuStudio.
On my A1 mini machine, printing can be restored using this method. 
[image]

Next, I will try to reproduce this issue on x1c

bam-winterfell, 2024-01-29:
@GirlPurple 
I have used the latest firmware version and BambuStudio version and have not reproduced your issue. Please try updating the printer firmware version and BambuStudio version. If there are still issues with restoring printing, please reply to me.

GirlPurple, 2024-01-30:
@bam-winterfell - 

According to what I can find online, I have both the latest firmware and the latest software version, as previously confirmed to you, and I was already using both when these failures occured. 

You shared the image of how to resume printing on your A1 before and I expalained then that my views DID NOT have a resume button or a play button - both screens only offered me a pause button, which did not respond to being clicked. No option but cancel worked. 

I asked you what specifcally you wanted me to record and you did not give me a clear answer on that. 

It feels like we are are talking in circles with me saying "it fails, and here's how" and you saying "but mine doesn't fail" and then telling me to check things I've already checked and/or told you don't work. 

Would you like me to send you the file I am trying to print w/ the layer 2 inserted pause so that you can try it on your X1 C and see if it fails for you?

bam-winterfell, 2024-01-30:
@GirlPurple 
If you are willing to send us the 3mf, that would be great. Also, please attach the printer log. You can refer to this wiki.[https://wiki.bambulab.com/en/x1/troubleshooting/export-failed-print-log-to-sd-card](url)
My email is hu.wang@bambulab.com
Thank you for your cooperation

s00500, 2024-02-19:
Hi there, I am having the very same issue, I have reported my logs and 3mf files as a bambu lab support ticket, would you like me to send this to you ?

I also have one file that seemed to work once.... it was sliced for me by a friend who seems to not have the issue...
Do you want me to send you the info too  via email ? (300mb logs are quite large...)

s00500, 2024-03-02:
Still having this issue, any news here ?

bam-winterfell, 2024-03-04:
hi@s00500 
I'm sorry for taking so long to reply to you. May I ask what is your printer model, firmware version, and studio version number.Do you also add a pause command to the gcode file, but there is no button to resume printing on the printer screen after pausing

s00500, 2024-03-04:
Hi @bam-winterfell 

Printer: X1C Combo
Firmware Version: 01.07.02.00 (Latest)

Yes I add a pause into gcode (at layer 2 specifically) and then get no resume button in most of the cases. Sometimes I get lucky and I get the resume button, but I could not yet figure out a pattern when it happens...

I am using latest Bambu Studio (1.8.4.51 on macOS) as well

s00500, 2024-08-06:
NOT inactive 

...and still a problem, but unfortunately impossible to reach people at bambu lab support, I have sent multiple rounds of logs and supplied video of how to recreate to them... al I get is yet another request for logs after 3 rounds... not sure how to proceed here

keyoss, 2024-08-06:
same!This is a blantant fail from bambulab!! i was looking out for changes and bugfixes each and every release: they simply adress nonesense bugs.i guess they are not capable of handling the slicercode where bambu studio relys on ;)--Gesendet mit der GMX Mail App

s00500, 2024-08-06:
Kein Grund unhöflich zu werden ;-)

I am very sure that this is a bug in the printer firmware, likely a race condition with the other checks that happen after layer one, like AI inspection.... but I cant seem to reach somebody in firmware team about this...

@SaltWei @bam-winterfell anything I can do to have you replicate this here ?

lanewei120, 2024-08-07:
@s00500 @keyoss 
I am sorry bam-winterfell returned to school serveral months ago
marked as inactive just mean this issue is not active for 3 months, and we will check this issue also

@walterwongbbl please check this issue

walterwongbbl, 2024-08-07:
Hi @s00500 
May I ask how you inserted the pause? Can you restore the scene.

SaltWei, 2024-08-07:
> Hi @s00500 May I ask how you inserted the pause? Can you restore the scene.

@walterwongbbl I guess by this
[image]

s00500, 2024-08-07:
@walterwongbbl yes, exactly like @SaltWei shows in the screenshot. Make sure to use exactly layer 2, I think the problem is related to that, I have not had issues with pauses on other layers yet

s00500, 2024-08-14:
@walterwongbbl any news here ? shall I resend my video for replication to someone ?

Kayu84, 2024-10-26:
X1C V1.08.02 still a Problem?!
added a pause on over bambu studio (1.9.7.52)
cannot resume the print. THIS is terrible...
HOW can you destroy my 40h print!!!

WHAT to do now? any ideas to resume manual?

s00500, 2024-10-26:
@Kayu84 try using a powercycle and recovery. If that does not work since you have a exact layer where it stops I would recommend printing a new print from exactly there and then gluing together. Likely your safest option

@walterwongbbl @SaltWei @bam-winterfell Any chance somebody can take a look at this again? It is very very frustrating that this always ends up getting forgotton, I have offered detailed help for replication, gathered logs multiple time, recorded a 6 min video showing the process and also sent multiple bambu studio files.

All that happened on official support. Can ANYONE here tell me where I can get a contact to help fix this issue once and for all?

Kayu84, 2024-10-27:
> @Kayu84 try using a powercycle and recovery. If that does not work since you have a exact layer where it stops I would recommend printing a new print from exactly there and then gluing together. Likely your safest option
> 

Thank you, found the same solution on reddit and it works. 🙄 and yeah, frustrating that an official feature didn't worked very well. 😵‍💫

Krakonos, 2024-11-29:
@bam-winterfell 

I just hit this issue on X1C. Pause after layer 2. I'll do more repro tests on a smaller model later to see if layer 2 is indeed the problem.

I managed to work around it by powercycling the printer and use recovery as suggested above. However, i see the issue is twofold:

1. The printer does not show the resume button in any dialog. This makes it impossible to resume from the device itself.
2. Slicer does not show the resume button either. I found out if I restart the slicer, a dialog pops up the printer is paused with a "Resume" button - however, clicking it does not work.

I don't have enough information to check if this is a bug in the printer or slicer - but my guess is there is an issue in both. Please have a look and fix the issue on the otherwise awesome printer! There are many similar tickets here, some of them even closed, many of them are year old or more.

Philipp150, 2026-11-01:
[image]

I installed a older Firmware on the Printer and it works again.

s00500, 2026-11-02:
what version ? are you dong layer 2 pause ?