bambulab/BambuStudio #2861: Annoying slicer tips - no way to close that immediately and most importantly no way to fully disable it

arekm, 2023-11-09:
### Bambu Studio Version

1.08.00.03

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 14.1.1

### Printer

Doesn't matter

### How to reproduce

1. Slice anything
2. Annoying things pops up

### Actual results

Big useless (for me) popup that cannot be disabled or even immediately closed.

[image]

### Expected results

Two things:
1) ability to close that tip immediately
2) ability to turn off these annoying things (that cover quite a lot of slicer space)

### Project file & Debug log uploads

Screenshot attached earlier.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DokuKaefer, 2023-11-09:
Same problem

lanewei120, 2023-11-10:
we will optimize this logic 

currently, you can close this tips when slicing a bigger model, which will be also valid for all models
[image]

kr0n1k01, 2023-11-21:
When will we be able to close these horribly obtrusive tips it stays on screen consistently. Why is there not a option to disable tips I seriously doubt users who are on betas want to see the tips pop up.

0rphon, 2023-11-22:
+1 to this. These tips could be useful to some people on their first few times printing, but are incredibly annoying to everyone else.

b00lduck, 2023-12-06:
Please add an option to disable this permanently. Precious screen space is wasted.

icheered, 2023-12-06:
These tips are really annoying, is there really no way to turn them off?

SaltWei, 2024-01-17:
Use latest version and if can be collapsed.

closed 2024-01-17 by SaltWei

kr0n1k01, 2024-01-17:
> Use latest version and it can be collapsed.

Sorry this does nothing to address the issue as even collapsing it reappears and takes up screen space. Still don't understand why there is no checkbox to show or hide tips.