bambulab/BambuStudio #2862: 1.8.0 Custom filaments not avilable for calibration or in AMS selection

capsel22, 2023-11-09:
### Bambu Studio Version

1.8.0

### Where is the application from?

Bambu Lab github releases

### OS version

Win11

### Printer

P1S

### How to reproduce

1. Create new custom filament
2. Go to Calibrate and select manual calibration
3. Custom filament not available
4. Go to Device and edit AMS filament
5. Custom filament not available

### Actual results

Created filament using custom filaments 
[image]

not available for calibration or AMS
[image]
[image]

### Expected results

To be able to do the above steps.

If calibration or assigning to AMS is not supported i fail to understand what's the benefit of doing it the Custom Filament way over just saving filament with a different name.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-11-10:
Hello @capsel22 

this also needs a firmware update
the printer team will release a public Beta version soon

rashuge, 2023-11-15:
Will this also be available for P1P/P1S and AMS?  I have the same issue.  I can create and configure the perfect filament profile but cannot actually calibrate or select in the AMS.  Very frustrating.

walterwongbbl, 2023-11-20:
@rashuge 
After P1P/P1S support, we will release a new configuration file. With the update, you can use Custom fillments.

closed 2023-11-24 by walterwongbbl