bambulab/BambuStudio #2863: Bambu Studio needs to be re-started when stopping a print at the printer

Supersprint, 2023-11-09:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Pro V 22H2 installed 24/10/2022 Build 22621.2506 Experience Pack 1000.22677.1000.0

### Additional system information

Device name	DESKTOP-LVU2IAO
Processor	Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz   1.50 GHz
Installed RAM	32.0 GB (31.8 GB usable)
Device ID	285AB8AC-EF03-4718-8E3F-EF8833C00401
Product ID	00330-52480-54045-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

### Printer

Bambu Lab X1Carbon Combo

### How to reproduce

When stopping a print at the touch screen on the printer, Bambu Studio does not recognise the print has stopped. When you go to the device tab, it shows the progress frozen at the point where the print was stopped. If you then click STOP on Bambu Studio, nothing happens. If you try to re-start the print, it shows the printer as being busy, so you can't start another print unless you close and re-open Bambu Studio. 

### Actual results

Please see above

### Expected results

When the print is stopped at the printer, Bambu Studio needs to recognise that the print has been stopped.

### Project file & Debug log uploads

[debug_network_Fri_Nov_10_09_18_08.zip](https://github.com/bambulab/BambuStudio/files/13313734/debug_network_Fri_Nov_10_09_18_08.zip)
[Adrian Dash Shield 2 v1 Plate.zip](https://github.com/bambulab/BambuStudio/files/13313736/Adrian.Dash.Shield.2.v1.Plate.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-11-14:
Hi. @Supersprint 
Can you provide the LOG file at the time of the incident?

walterwongbbl, 2023-11-14:
If possible, it would be best to provide relevant videos. This can help us find the cause faster.

walterwongbbl, 2023-11-14:
Hi. @Supersprint 
By analyzing the LOG file you provided, we didn't find the cause. However, it may be related to our message publishing, possibly resulting in the loss of some critical messages. You can try using our 1.8.0.0 version, where we have made some optimizations. Thank you very much for your support.

closed 2024-01-18 by walterwongbbl