bambulab/BambuStudio #2864: Time delay start

Supersprint, 2023-11-09:
It would be helpful to have a time delay start feature for the printer. This would mean a print could be started when everyone is asleep, as it's quite noisy and hard to get to sleep whilst running.

walterwongbbl, 2023-11-10:
Hi。@Supersprint 
Do you mean delaying printing?

Supersprint, 2023-11-10:
Yes, delay printing

Regards Jon Newell

ATTENTION: This email may contain private and confidential information intended only for the use of the original recipient named above. If you are not the intended recipient of this email you are hereby notified that any use, dissemination, distribution or reproduction of this message is prohibited and any such action can contravene the NSW Privacy and Personal Information Protection Act 1998. If you have received this message in error please notify the sender immediately by return email and then permanently delete the original email including the notification of receipt of the email to the sender.

ben-h-a, 2023-12-15:
Also think this would be a useful feature, it can be distracting even with the printer in another room.

jkrauska, 2025-12-02:
Agreed -- my power costs are much higher (almost double)  until after midnight.  If I could "time shift" and schedule a print to start at 1am, it would save me a lot in power bills.