bambulab/BambuStudio #2865: Unable to disable Daily Tip

Candyffm, 2023-11-10:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Printer

P1P

### How to reproduce

After every successful slice, a massive daily tip window appears and blocks about 10% of the screen. There is no option to disable these messages.

### Actual results

[image]

[image]

### Expected results

At least give us an option to disable that feature

### Project file & Debug log uploads

no specific project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LiZ-Li-BBL, 2023-11-13:
Thank you for your suggestion. We are considering to improve the experience of this feature in the next release.

etazelaar, 2023-11-13:
+1 on Bambu Studio 1.8.0.57. Please allow us to disable daily tips. I just updated all my computers, 4, yesterday to 1.8.0.57 plugin version 1.8.00.03 build time 20231109-141115.

mltriebe, 2023-11-17:
+1, PLEASE

GuzziRaz, 2023-11-28:
Still a problem in 1.8.0.62. We can now close it immediately, but on small-ish screens (I run 2880x1800) the close button is behind the slicing results box. So I have to minimize that box before I can close the annoying tip.

I really think Daily Tips are fine, but they are usually shown when you start a program, not during work in it. This is like MS Office notorious animated paper clip that everyone really hated.

I suggest that you show the tip at program start only, AND add an option to disable it.

tikard, 2023-12-08:
I agree with this mod and could you make option to not open Color Scheme info screen every time it goes to preview.

NeoBlack, 2023-12-15:
+1 please disable this annoying popup on every slicing. Please give us at least the option to disable it.

GuzziRaz, 2023-12-15:
Just collapse it and it will remain collapsed

LiZ-Li-BBL, 2023-12-19:
[image]
You can fold the dailytips window now, and once you fold it, it stays folded.
[image]

copc, 2023-12-22:
Please this is annoying, thanks

TennisRunner, 2023-12-25:
Why is this even a thing. Do you really think its a great a idea to add forced nag screens?
Nobody used this and thought that is really annoying?
Are the developers really that disconnected from their own product?

bradanlane, 2024-01-17:
the "tip" message is annoying. It is present in 1.8.4.51. please allow users to disable this "tips" after the slicing.

Burmuley, 2024-03-22:
Please make an option to disable this annoying popups! PLEASE! ASAP....

dguimaraes88, 2024-04-17:
Hi People!
You can solve this with this method:
Press Windows Key + R -> %APPDATA%\BambuStudio -> Enter

Edit File: BambuStudio.conf

Replace this value from: 
"show_daily_tips": true,
to:
"show_daily_tips": false,

And....thats it!

[image]

rodpgh, 2024-05-03:
how do to remove on a Mac?

FWeinb, 2024-05-04:
@rodpgh on macOS the path to the config folder of BambuStudio is `~/Library/Application Support/BambuStudio`. Open Finder and press ⇧⌘G and past that path, you will find the `BambuStudio.conf` there.

rodpgh, 2024-05-07:
@FWeinb Thank you .. it worked :)

fluffy-critter, 2024-09-26:
I'm still having this on Bambu Studio 1.9.7.52, and it feels like it had gone away for a while but now it's back with a vengeance.

TJC, 2025-08-24:
I'm on version 2.2.0.85 and I still can't find any option in the Preferences to disable the Daily Tips.