bambulab/BambuStudio #2866: Query Current Bed Temp for Conditional G-code?

CCS86, 2023-11-11:
In SS / PS with my Duet3D powered printer, I use conditional gcode to look at the current bed temp and decide whether it is cold and needs a longer heat soak, or if it is already hot and can start a print immediately:

```
{"if heat.heaters[0].current < "}[first_layer_bed_temperature]-1.5
{"	"}M190 R[first_layer_bed_temperature]               ; Wait for cold bed
{"	"}G4 S440                                                             ; Cold Pause
{"else "}
{"	"}M190 R[first_layer_bed_temperature]               ; Wait for hot bed
{"	"}G4 S1                                                                ; Hot Pause
```

Is there is a system variable in the Bambu firmware that we can query to get the current bed temp, and support for conditional operations like this?

CCS86, 2023-12-01:
???

CCS86, 2024-02-08:
Hello?

CCS86, 2024-03-15:
@QingZhangBambu ?

QingZhangBambu, 2024-03-18:
I will ask the concerned staff about that.

QingZhangBambu, 2024-03-18:
@CCS86 firmware is not support that.