bambulab/BambuStudio #2869: When you go to print using an AMS and multi materials, the message has a spelling mistake

DarrenLynch, 2023-11-11:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Mac

### Printer

N/a

### How to reproduce

1. Start a print with multi-material with an AMS
2. A dialog pops up, something along the lines of, "AMS material mappings established, you **car** click on the materials..." Car should be can

### Actual results

car

### Expected results

can

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-11-14:
Hi. @DarrenLynch 
Can you provide screenshots or videos? We did not see the word 'car' in the translated document

walterwongbbl, 2023-11-14:
Hi. @DarrenLynch 
Please use our latest version 1.8.0.0 and verify if the issue has been resolved. Thank you very much for your support.

closed 2023-11-24 by walterwongbbl