bambulab/BambuStudio #2870: Fault when cutting a frame shaped object 90 degrees over X- or Y-axis (1.8.0.57)

coat64, 2023-11-11:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

AMD 5700G
64 GB Mem

### Printer

X1C

### How to reproduce

Load a simple and correct frame shaped object (stl)
Cut it over the X-or Y-axis at 90 or 270 degrees

[image]
[image]
[image]
[image]

cutting at a different angle other then square seems to work correctly 

### Actual results

The 2 resulting shapes are faulty (non manifold)
slicing gives a faulty result
repairing the part does not work properly
(Rightclick on a part and "fix model" does work)

### Expected results

Cut correctly

### Project file & Debug log uploads

The project not cut yet
[test_cut_frame.zip](https://github.com/bambulab/BambuStudio/files/13325791/test_cut_frame.zip)
The stl of the simple frame
[testvb_stl.zip](https://github.com/bambulab/BambuStudio/files/13325793/testvb_stl.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

coat64, 2023-11-11:
When you move the cut, it seems the problem disappear.

Haidiye00, 2023-11-13:
Model cutting appears popular edge, please first through this through the model repair, later we will fix the problem
[image]

closed 2024-01-16 by Haidiye00