bambulab/BambuStudio #2871: Top window bar disappear on mac when maximising or moving the BambuStudio Window

Xuheir, 2023-11-11:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS Sonoma 14.0

### Additional system information

Apple M2 Ultra
128 GB

### Printer

Bambu X1C

### How to reproduce

open the App  
maximise or move the window to the side
the top bar disappears, Can't after move or minimize the app
the only way is to quit and then open again it agian.

### Actual results

the app starts normally and works but if you move the window to the side or maximize it the top bar disappears.

### Expected results

the top window bar should stay as in any other mac app or window.
i am talking about the bar with the three dots to quit, minimize or maximize 
Mac

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-11-14:
Hi @Xuheir 
[image]
Have you ever set the top title bar of the system? Please check if the settings have caused the system to not display the restore button. This is a recording of my operation on macOS 14.0. Please confirm if it matches your operation.

Xuheir, 2023-11-14:
Hi,
thank you for looking into this,
I got the same effect as you showed in your screen recording when
maximizing/minimizing.
I found out the reason for the issue I have in my system,
it's the BetterSnapTool software causing the (top bar on BambuLab only )
when I snap the window in any direction to disappear.
solution I have created a shortcut in BetterSnapTool to restore the window
of BambuLab to its previous size and position.

Steven

-- 

Steven Serhan

________________________________________________________
The information contained in this e-mail message and any accompanying files
is or may be confidential. If you are not the intended recipient, any use,
dissemination, reliance, forwarding, printing or copying of this e-mail or
any attached files is unauthorized. This e-mail is subject to copyright. No
part of it should be reproduced, adapted or communicated without the
written consent of the copyright owner. If you have received this e-mail in
error please advise the sender immediately by return e-mail or telephone
and delete all copies. Steven Serhan does not guarantee the accuracy or
completeness of any information contained in this e-mail or attached files.
Internet communications are not secure, therefore Steven Serhan does not
accept legal responsibility for the contents of this message or attached
files.

walterwongbbl, 2023-11-20:
@Xuheir 
good job!

closed 2023-11-20 by walterwongbbl