bambulab/BambuStudio #2872: Ironing inset

TestApp90, 2023-11-11:
Hello,

could you please implement the Ironing Inset feature like in Cura? It's hard to combine the correct amount of ironing flow with clean edges right now. The ironing inset can be used to set the distance to keep from the edges of the model.

Like here:
[image]
[image]

Thank you.

thinkingalaud, 2023-11-15:
+1, would love to have this feature. On curved edges, a step pattern appears on the outside of the boundary of the model after ironing

[image]

SaltWei, 2023-11-20:
@QingZhangBambu Please support this in 1.9 version

thinkingalaud, 2024-05-15:
Any update on this @QingZhangBambu ? I noticed 1.9 was released recently, but I didn't see this in the release notes.

QingZhangBambu, 2024-05-16:
@thinkingalaud Sorry, it been delayed.  I will add it to 1.10

PeteCondliffe, 2024-06-22:
I can't find this anywhere did it ever get added? If so would someone be kind enough to point me in the right direction.
Thanks

PeteCondliffe, 2024-06-22:
Does this exist now? if so please could you point me to where? Thanks

SaltWei, 2024-06-24:
> I can't find this anywhere did it ever get added? If so would someone be kind enough to point me in the right direction. Thanks

Will be added at 1.10 version @QingZhangBambu

PeteCondliffe, 2024-06-24:
Apologies i read it as 1.1 not 1.10

thinkingalaud, 2024-11-06:
Just saw this in the docs - it's happening!!

[image]

closed 2025-05-30 by QingZhangBambu